makerbase-mks/MKS-SBASE #5: License

GearheadRed, 2016-03-16:
What license does this board and the data contained in this repo fall under? The information seems to be missing from the repository and would be nice to know what it is.

lordofhyphens, 2016-03-22:
As far as I've been told by sellers it's closed source.

GearheadRed, 2016-03-22:
Thats fine, but I would like to get a response directly from the manufacturer/designer. If possible include the license within the repo for clarity to others finding the repo.

lordofhyphens, 2016-03-22:
Yeah, probably never going to get it though :(

Something to remember is that if there is no license and it explicitly
hasn't been released to public domain, there's no right to use it.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

GearheadRed, 2016-03-23:
I understand how the lack of license rule/concept works. That does not mean I cannot ask for clarification on the issue. Im just going to leave this here and wait to see if the repo owner comments.

SigmazGFX, 2016-07-25:
Hows that working out for ya?

arthurwolf, 2016-07-25:
The MKS guys said on the reprap forums that this is not open-source and you
can't use the files ... Are also well known for being unhelpful to
customers... I wouldnt expect them to do much github communication.

GearheadRed, 2016-07-25:
So I get the opinions and all that. You are all entitled to that. But
really what's wrong with asking for an "official" position from the git
owner? If they don't respond that's their prerogatives. I realize the
position the community has with them, I'm not asking for what the community
thinks here.

arthurwolf, 2016-07-25:
Sure I'm not saying there is any problem with asking. Was just reporting
what they said elsewhere on this exact topic, and what I think about it.

SigmazGFX, 2016-07-25:
I can't even imagine how infuriating this must be for you arthur.

closed 2017-02-06 by makerbase-mks

arthurwolf, 2017-02-06:
Ahaha :)