makerbase-mks/MKS-SBASE #7: Does not support multiple power supplies

lordofhyphens, 2016-03-24:
The MOSFETs all appear to be between the heater/fan terminals and VCC, which removes the ability to bypass the main power input so that a second power source may be used (24V heater, 12V stepper motors, or 12V fans and 24V stepper drivers). 

Most other control boards put the MOSFETs on the GND path instead.

makerbase-mks, 2017-02-06:
yes mks sbase only support one power supplies.

closed 2017-02-06 by makerbase-mks

evgs, 2017-03-28:
All mosfets on the SBASE are N-channel, sitting on the ground path. So board may work with both 12V and 24V if load + terminal will be connected to appropriate power source directly. Also it may be nessesary to unsolder some on-board LEDS, if board is powered 12V and load is 24V (else leds will be reverse-biased by 12V)

evgs, 2017-03-28:
@lordofhyphens you may browse trough git history and dig out board sch ;)