makerbase-mks/MKS-SBASE #12: MKS Smini PINs

DaGov, 2016-08-05:
Need MKS Smini PIN diagram for connection. Or create new branch for this board with info.

brunnels, 2017-01-14:
The best I found was this https://img.alicdn.com/imgextra/i3/12778391/TB2S7lguXXXXXXEXpXXXXXXXXXX_!!12778391.png

The problem is I'm assuming the 16/128 jumper headers is for switching between 16/128 microstepping but is there any option in between?  Also there is the 4 position switch that is labeled as "SDCard Switch" but no documentation on that.

makerbase-mks, 2017-02-06:
mks smini only have 16 and 128, can't use  other microstep.
SDCard Switch is used to set if use sdcard on board and sdcard on 12864lcd at the same time or only use sdcard on board . On normal time, it is not useful

closed 2017-02-06 by makerbase-mks