makerbase-mks/MKS-SBASE #14: What is DRV8825 decay mode used by board?

egorse, 2016-09-26:


pixel3design-hub, 2016-10-04:
If you mean Fast Decay Mode, it's an internal function of the DRV8825. It's a good function badly implemented, even in the original Smoothieware Hardware. It needs diodes in reverse parallel on two channels of the output to stop motor making terrible noise in 1/32 step mode, just because current phase control is generating harmonics and sub harmonics from the 30KHz pulse signal that chip is working on.

Kinda of...

egorse, 2016-10-04:
I am on pin 19 - is that low, open or high?

pixel3design-hub, 2016-10-04:
Low = slow decay, open = mixed decay,
high = fast decay.
Internal pulldown and pullup.

egorse, 2016-10-04:
I can see that from datasheet - but whats on the PCB/assembly?

tiaanv, 2016-11-23:
I second the request.  it the PCB wired to put the DRV8825 in fast decay mode or is is mixed mode?  Would not be able to see unless I remove the heatsink which I would not like to do.  Listening to it in operation, I think it might be in Mixed mode.  Please confirm.

Steamerzone, 2017-01-02:
As far as I can measure it, it's running in mixed decay mode. (decay pin is open)

makerbase-mks, 2017-02-06:
MKS Sbase's drv8825 all use mixed decay mode.

closed 2017-02-06 by makerbase-mks

feket663, 2017-02-09:
Can I use this board with 24 power supply and relativly high inductance stepper motors with low current? Need enable fast decay mode?