makerbase-mks/MKS-SBASE #15: Poor print quality with MKS Sbase v1.3

renzaaa, 2017-01-12:
I was printed a lot of things with different settings, but layers of the outline is a jaggy.
I don't understand why, because with another board (marlin fw) the outlines quality is excellent.
Something like this: **https://www.simplify3d.com/wp-content/uploads/2015/09/Lines-On-Side-Of-Print.jpg**
My printer is cartesian style.
Can someone help me?
This is my cofiguration:

# NOTE Lines must not exceed 132 characters
## Robot module configurations : general handling of movement G-codes and slicing into moves
default_feed_rate                            4000             # Default rate ( mm/minute ) for G1/G2/G3 moves
default_seek_rate                            4000             # Default rate ( mm/minute ) for G0 moves
mm_per_arc_segment                           0.0              # Fixed length for line segments that divide arcs 0 to disable
mm_max_arc_error                             0.01             # The maximum error for line segments that divide arcs 0 to disable
                                                              # note it is invalid for both the above be 0
                                                              # if both are used, will use largest segment length based on radius
#mm_per_line_segment                          5                # Lines can be cut into segments ( not usefull with cartesian
                                                              # coordinates robots ).

# Arm solution configuration : Cartesian robot. Translates mm positions into stepper positions
alpha_steps_per_mm                           160.36               # Steps per mm for alpha stepper
beta_steps_per_mm                            160.36               # Steps per mm for beta stepper
gamma_steps_per_mm                           800                  # Steps per mm for gamma stepper

# Planner module configuration : Look-ahead and acceleration configuration
planner_queue_size                           32               # DO NOT CHANGE THIS UNLESS YOU KNOW EXACTLY WHAT YOU ARE DOING
acceleration                                 3000             # Acceleration in mm/second/second.
#z_acceleration                              500              # Acceleration for Z only moves in mm/s^2, 0 uses acceleration which is the default. DO NOT SET ON A DELTA
junction_deviation                           0.05             # Similar to the old "max_jerk", in millimeters,
                                                              # see https://github.com/grbl/grbl/blob/master/planner.c
                                                              # and https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8
                                                              # Lower values mean being more careful, higher values means being
                                                              # faster and have more jerk
#z_junction_deviation                        0.0              # for Z only moves, -1 uses junction_deviation, zero disables junction_deviation on z moves DO NOT SET ON A DELTA
#minimum_planner_speed                       0.0              # sets the minimum planner speed in mm/sec

# Stepper module configuration
microseconds_per_step_pulse                  1                # Duration of step pulses to stepper drivers, in microseconds
base_stepping_frequency                      100000           # Base frequency for stepping

# Cartesian axis speed limits
x_axis_max_speed                             12000            # mm/min
y_axis_max_speed                             12000            # mm/min
z_axis_max_speed                             300              # mm/min

# Stepper module pins ( ports, and pin numbers, appending "!" to the number will invert a pin )
alpha_step_pin                               2.0              # Pin for alpha stepper step signal
alpha_dir_pin                                0.5!              # Pin for alpha stepper direction
alpha_en_pin                                 0.4              # Pin for alpha enable pin
alpha_current                                0.8              # X stepper motor current
alpha_max_rate                               12000.0          # mm/min

beta_step_pin                                2.1              # Pin for beta stepper step signal
beta_dir_pin                                 0.11!             # Pin for beta stepper direction
beta_en_pin                                  0.10             # Pin for beta enable
beta_current                                 0.8              # Y stepper motor current
beta_max_rate                                12000.0          # mm/min

gamma_step_pin                               2.2              # Pin for gamma stepper step signal
gamma_dir_pin                                0.20!             # Pin for gamma stepper direction
gamma_en_pin                                 0.19             # Pin for gamma enable
gamma_current                                0.8              # Z stepper motor current
gamma_max_rate                               300.0            # mm/min

## System configuration
# Serial communications configuration ( baud rate defaults to 9600 if undefined )
uart0.baud_rate                              115200           # Baud rate for the default hardware serial port
second_usb_serial_enable                     false            # This enables a second usb serial port (to have both pronterface
                                                              # and a terminal connected)
#leds_disable                                true             # disable using leds after config loaded
#play_led_disable                            true             # disable the play led
[... 271 lines cut ...]
panel.gamma_jog_feedrate                     200               # z jogging feedrate in mm/min

panel.hotend_temperature                     200               # temp to set hotend when preheat is selected
panel.bed_temperature                        50                # temp to set bed when preheat is selected

## Custom menus : Example of a custom menu entry, which will show up in the Custom entry.
# NOTE _ gets converted to space in the menu and commands, | is used to separate multiple commands
custom_menu.power_on.enable                true              #
custom_menu.power_on.name                  Power_on          #
custom_menu.power_on.command               M80               #

custom_menu.power_off.enable               true              #
custom_menu.power_off.name                 Power_off         #
custom_menu.power_off.command              M81               #

## Network settings
network.enable                               false            # enable the ethernet network services
network.webserver.enable                     true             # enable the webserver
network.telnet.enable                        true             # enable the telnet server
network.ip_address                           auto             # use dhcp to get ip address
# uncomment the 3 below to manually setup ip address
#network.ip_address                           192.168.3.222    # the IP address
#network.ip_mask                              255.255.255.0    # the ip mask
#network.ip_gateway                           192.168.3.1      # the gateway address
#network.mac_override                         xx.xx.xx.xx.xx.xx  # override the mac address, only do this if you have a conflict

pixel3design-hub, 2017-01-16:
Hi,
did you run the AutoPID?
This seems to me as inconsistent heating problem.
PID in Marlin is different from Smoothieware PID. Values does not match.
Run:
M303 E0 S200

Wait until it's finished. PID settings are now writtened into SD card and memory. Just do M500 and you are ready to go.

renzaaa, 2017-01-16:
No, I don`t.
Thanks,
I will try.

pixel3design-hub, 2017-01-18:
Let Us know if you have better quality now. Thank You

renzaaa, 2017-01-18:
Nothing changed. 
Links for my printed objects:
https://drive.google.com/open?id=0B-ICy1pydz_DNVhYWDAxbVg3VXM
https://drive.google.com/open?id=0B-ICy1pydz_Da0J5U2xRbkFFQTA

pixel3design-hub, 2017-01-27:
Hi...
Another issue could be driven by insufficient extruder power. Did you have a direct drive or a geared one?
Try:
. take the filament out of the hotend, cold-extrude (M302 followed by a G1 E50 F100) and try to retain the filament. If a low force makes the motor stuttering, then there is your problem. Ramp up the current of the extruder (delta_current 1.0). Anyway 0.8 is a little low for the extruder.

. get a lower speed during print. If you go at high speed, you need more heat, because the hotent heat is constantly ripped off by the heat transfer between hotend and filament. So try to increase the temp or low the speed.

Hope it'll work. 
Cheers

renzaaa, 2017-01-27:
Hi..
I have direct drive extruder.
I was try a slower speed and was changes the temperature, also I was try a different versions of the smoothieware and slicer softwares. 
OK, I will try.
Thanks

renzaaa, 2017-01-29:
Nothing changed... 
I was try replace E0 with E1 and still with a same problem.

pixel3design-hub, 2017-02-02:
Dunnow dude... Last option (at least for me) it's to try to print an high tower and searching in it for patterns. If you can find it, it could be possibly the threaded rod

lordofhyphens, 2017-02-06:
@renzaaa 
The SBASE uses DRV8825 drivers, you probably want to confirm that this isn't a problem.

https://www.reddit.com/r/3Dprinting/comments/5kful2/so_ive_recently_discovered_that_missing_steps/
http://cabristor.blogspot.com/2015/02/drv8825-missing-steps.html

robustini, 2017-04-19:
See my post here, same result:

https://groups.google.com/forum/#!topic/smoothieware-support/jlJKE_zbgno

JanN-H, 2017-11-06:
I know, it's an old thread, but my first idea was: alpha_steps_per_mm and beta_steps_per_mm are not plain values. Try to set them to 160.

closed 2020-06-06 by makerbase-mks