makerbase-mks/MKS-SBASE #16: Nema 14 stepper motor set up

panamike, 2017-01-19:
I want to use a Nema 14 stepper motor to drive my extruder. But skips no mater what settings i use.
I am using Pololu 200 Steps/Rev, 35×28mm, 10V, 0.5 A/Phase

Settings are:

[[extruder.hotend.enable]] [[true]]
[[extruder.hotend.steps_per_mm]] [[1067]]
[[extruder.hotend.default_feed_rate]] [[600]]
[[extruder.hotend.acceleration]] [[5]]
[[extruder.hotend.max_speed]] [[60]]

[[extruder.hotend.step_pin]] [[2.3 ]]
[[extruder.hotend.dir_pin]] [[0.22]]
[[extruder.hotend.en_pin]] [[0.21]]

[[delta_current]] [[0.9]]

makerbase-mks, 2017-02-07:
please use the step motor on x/y/z axis and confirm it can work.

or maybe [[extruder.hotend.steps_per_mm]] [[1067]] too big,change to a small number and test.

dieideeistgut, 2017-02-07:
Hey there - using Nema14s here all around. I have a geared extruder but my config is the following:

```
extruder.hotend.enable                          true             # Whether to activate the extruder module at all. All configuration is ignored if false
extruder.hotend.steps_per_mm                    140              # Steps per mm for extruder stepper
extruder.hotend.default_feed_rate               600              # Default rate ( mm/minute ) for moves where only the extruder moves
extruder.hotend.acceleration                    500              # Acceleration for the stepper motor mm/sec²
extruder.hotend.max_speed                       50               # mm/s`
```
---
```
delta_current                                0.6750              # First extruder stepper motor current
```

Also keep in mind that there a Nema14s with ways different specs out there. Your motor may skip because it's getting too hot as you're driving it with too much juice ;) The smaller Nema14s quite like to lose magnetic force when they get hotter so they start to skip. I'm running the Nema14s rated with 0.8A - you should maybe recheck your specific motor. I also can recomend running at about 80% of the max-current and even adding a fan to the motor helps a lot. Other than that Nema14s are brilliant ;)

Hope this helps.

closed 2020-06-06 by makerbase-mks