makerbase-mks/MKS-SBASE #17: Hotend2 doesn't work.

ecagatay, 2017-01-21:
Hi,

I have the SBASE 1.3 and I have it working fine for the first hotend. For the life of me, I can't get the setup for the second hotend to work. Have you (or anyone on this thread) got dual/multi-extrusion working on this board? My board's pins work fine, but I can't work hotend2.

I have a temperature reading of 38.5C on the second termistor input (regardless if anything is connected). The power mosfet LED for the E2 doesn't light up either if I try to turn it on (and doesn't set a new target temp for it). Anyone having the same problem? Any examples of a working setup?

Please help me.
Thanks.

makerbase-mks, 2017-02-07:
please conform the config is ok.

pixel3design-hub, 2017-02-13:
pls post your config.txt

rogerwilco3, 2017-09-05:
Hello make sure the M code is configured properly the code supplyed are 889 in a std setting change this to 104 and 109.in the std file the link.to.tool is absent i only changed the right m-code in the following line

#temperature_control.hotend2.thermistor_pin 0.25 # Pin for the thermistor to read
#temperature_control.hotend2.heater_pin 2.6 # Pin that controls the heater
#temperature_control.hotend2.thermistor RRRF100K # see http://smoothieware.org/temperaturecontrol#toc5
##temperature_control.hotend2.beta 4066 # or set the beta value
#temperature_control.hotend2.set_m_code 884 # change this to 104
#temperature_control.hotend2.set_and_wait_m_code 889 # change this to 109
#temperature_control.hotend2.designator T1 #
you will find now that the led at d12 will light up when Extruder 2 is heating up

closed 2020-04-02 by makerbase-mks