makerbase-mks/MKS-SBASE #20: High pitch noise from motor

v27jain, 2017-03-14:
I am using MKS SBASE 32-bit board with smoothieware. After connecting to power supply and homing the axis, a high frequency noise starts coming out from the motors. How can we reduce/remove the high pitch  noise generation ?

ahmetipkin, 2017-03-30:
Mine does these too. Dropping the current a little bit kinda alleviates it but never fixes it. I think it is the stepper drivers making that noise and have no solution for it.

MD3DDelta, 2017-04-02:
Hy , i dropped the curent to 0.7 and the noise is gone , or i don't hear it anymore .
nema 17 0,9degr motor installed.
default seting was 1.0 , at 0.8 still some noise. the motor got hot too, that's what they do not have to do.
when they get to hot the motors live quite shorter , and the pla where it is mounted can deform in time .
Mine motors now about 40derg celcius in full speed and stays around it ,i can touch them without burn my fingers. just try out , and keep your finger on the reset button for in case something goes wrong  .
with nema17 1,8degr motor i have not testet,

ahmetipkin, 2017-04-02:
I have the similar motors and yeah, I also use 0.7 :) I think 1.8 motors would be more quiet since they have less coil wraps in them but I didn't test it either. 

Happy printing fellas ;)

vader333, 2017-08-15:
After some research, it seems it's a "feature" of the DRV8825. I'd really prefer if they used LV8729's. It's not that much more expensive, and it doesn't have the annoying whine.

cunbal, 2018-10-03:
That is very annoying sound it is a pain to my ears! I cant live with that sound.
I'm running 1.8 A motors. Tried 12v-24v up/down currents with no go!

Looking for external drivers :/ Lost my motivation to work with that board.

closed 2020-06-06 by makerbase-mks