makerbase-mks/MKS-SBASE #21: SMINI doesnt flash

inzKrzys, 2017-03-22:
I try flashing firmware with 4 freshly formated to fat32 sd cards and its not working. File extension of firmware.bin stays the same.
(All of this cards works perfectly in other devices)
I try this on 3 win10 pc`s, with 32 and 64bit system and board doesnt show up in device menager.
I install windows drivers and nothing change.
I try entire procedure with and without external power source.

No matter what I do, led pattern stays the same:
- D5 and D6 (near mosfets) light up for 2 seconds and stays off for 2 seconds.
- D3 stays lit entire time
- D4 stays lit but barely visible

Is it possible that seller send board without bootloader?

inzKrzys, 2018-03-10:
Still no

pixel3design-hub, 2018-03-11:
It could be...

makerbase-mks, 2018-03-29:
@inzKrzys  Hvae you install the usb driver?

pixel3design-hub, 2018-04-05:
post a photo pls.
There's no D5 D6 near mosfets. I suppose you got a different board.
MKS-SBASE LEDS:

        D7  D4  D3  D2  D1
        R7  R4   R3  R2  R1
uSD

ETHERNET
                                                  MCU

USB

                      D14                         D13                           D12                            D10

inzKrzys, 2018-04-05:
@makerbase-mks 
Yes

@pixel3design-hub 
It is SMINI as it stands in title of thread. SMINI is the same as SBASE but without Ethernet and with less power outputs. Chinese manufacturer does'nt care to make separate chapter for this board and firmware is the same as in SBASE so I wrote here.
I have suspitions that board itself is broken but post cost to and from seller is worth same as board itself so meh ¯\\__(ツ)__/¯
[image]

pixel3design-hub, 2018-04-07:
Ok. On aux1 connector, TX and Rx to flash bootloader, close the boot jumper

ArchConMan, 2018-09-25:
> Ok. On aux1 connector, TX and Rx to flash bootloader, close the boot jumper
[image]
boot jumper is what? that? I have the board, not much info, even less of a clue! Thanks.

closed 2020-06-06 by makerbase-mks