makerbase-mks/MKS-SBASE #25: SPI 0 and 1 pins?

cruwaller, 2017-05-11:
How is the SPI0 and SPI1 connected? What pins should be used?

pixel3design-hub, 2017-05-12:
For?

cruwaller, 2017-05-12:
Really?
TMC2660, max31865 and display....

pixel3design-hub, 2017-05-12:
The TMC can be used with dir/step pins.
The MAX31xxx can be connected on the J7 pins (here for thermocouples).
Display... you got two 10 pins strips, one via serial AUX and another one via SPI...

Try to stay calm and chill. I'm not a SMOOTHIEWARE dev, nor a MKS one.
If you want more infos, buy a Smoothieboard and ask 'em.
Here we are, just peoples trying to help other peoples.

Cheers

cruwaller, 2017-05-12:
Yeah, i am :)
Just wondering the the SPI pins, cannot find SPI0 SCK pin (0.15). SPI1 overlaps with the LCD pins (reprap discount). 
This should be pretty easy question after all :) even for the closed hw :D

Smoothie is open source stuff and MKS should clearly define pins & capabilities at least :)

pixel3design-hub, 2017-05-13:
Sure they should.
But I'm using MKS for a long time and many devs complains for the lack of credits to smoothieware.
As you can see, nothing changed.
Chinese companies don't give a f..k of international laws. And it's a fact.
I really hate closed source, but Gov's should start to enforce OpenSource protection.
It would be easy to add some greets to Smoothie peoples. But as you can see...

Cheers

cruwaller, 2017-05-15:
Agree!
hope they release some info but i have my doubts :| 
I might need to find those by myself.

Thanks anyway!

lollandster, 2017-12-04:
I had the same issue and since this is the first result on google I guess I should share my findings here.
P0.14 isn't actually a pin on the LPC1769, it is obviously a typo in the sbase pinout diagram, that pin is the P0.15 SCK0 pin.

I have successfully connected a SSD1306 to the MKS sbase. (the smoothieware wiki doesn't mention it, but the panel.lcd value should be ssd1306_oled. I had to look in the source code to find it.)

cruwaller, 2017-12-27:
hah, I found the answer finally!
SPI is connected to EXP2 header pins and can be selected by J16:
- CS2 P0.28
- SCK P0.7
- MISO P0.8
- MOSI P0.9
CS pin can be any available GPIO pin.

arhi, 2018-03-16:
@cruwaller where did you see the description of what J16 does (looking for it myself)

waran14, 2018-06-14:
the SD card slot on my RRD Full graphics panel is not able to read any files. Does this have anything to do with the SPI's mentioned here ?

The display works fine as expected. It shows 2 options: ext and sd. But nothing is shown inside ext. I have tried 3 different cards until now

closed 2020-06-06 by makerbase-mks