makerbase-mks/MKS-SBASE #26: Printer stop when heating or printing.

stanrob, 2017-05-15:
Hi, 
Since a few days, my printer stop working during pre-print heating or after a few minutes of print. 
It is like a reboot of the board. I try changing the Firmware and the g-code file but in every configuration I have the same issue. 
Do you know why?

MD3DDelta, 2017-06-08:
Hi , i had same problemm , in mine case it was the temp sensor in the hotend , the wire was half broken , so the board got problems and resets its self. I repaired the wires, and earthed all metall from the printer .to aviod this again , this is also to prefend static-electic from you , when you touch the printer, 
With testing, the board gives me ,error halted temp 1," so it works fine now.
cooling the board helps too, 
hope for you that you find the

stanrob, 2017-06-08:
Hi, thanks for your answer. For now I change the heat cartridge for a 30W and every things works fine. 
I also thought to use a 24V power supply instead of a 12V but I didn't did it so I can't tell about it.

amelo81, 2017-07-22:
Hi normally i use sbase on my printers, just the new v1.3 i am getting problem that it keep stopping and running by it self.... can be after 5 min or 1 hour print... is not a gcode problem i am sure because running on another printer, normally stops for a few seconds many times on same print. Anyone already face this problem?

pixel3design-hub, 2017-07-24:
It's a common problem.
I resolved it by switching to the MASTER BRANCH of Smoothieware firmware, more stable than the DEV branch.
Removed ALL the comments from the GCode

AlexSkarbo, 2017-08-25:
Hello! I have a similar problem only hangs randomly, there is no dependency, I do not know what to do anymore ((The cable from MKS-TFT 32 to MKS-SBase v1.3 was shielded with aluminum tape, all the sensors and heaters work fine, the printer's metal case H-bot and it is grounded.When it hangs up, it stops reacting to any actions, buttons on the screen are pressed but the printer does not respond, does not execute commands on telnet (returns only the exclamation mark (!)).

stanrob, 2017-09-04:
Hi, so I try a few things to solve this and sometime when it append to me I just remove every gcode in my usb key and reput the current gcode on it and it is working again.
I figure out that it doesn't append on big print (when the first hour works the rest of the print will work to).

closed 2020-06-06 by makerbase-mks