makerbase-mks/MKS-SBASE #27: Second heater does not work. Design Flaw

ficklecycler, 2017-05-31:
Hello,

I have tried multiple different configuration settings but I cannot get the second heater to turn on. The thermistor associated with E2 does get a reading with the configuration I have, but I can't get any response from the heater. The LED D12 does NOT turn on.

Please, Makerbase, or anyone who has gotten this working, provide a config file that solves this issue. I saw that another user on the github didn't receive any help. I have seen this issue posted in several outlets now including the reprap forums and so far Makerbase has not proved that this board is designed in a way that allows the second heater to turn on.

Prove me wrong please, I'd love to have this board working and be able to recommend it to friends.

Thank you.
[config.txt](https://github.com/makerbase-mks/MKS-SBASE/files/1042285/config.txt)

pixel3design-hub, 2017-06-02:
Remove .link_to_tool on both hotend 1 & 2.
This disable the second heater because is declared as linked to the first. So thermistors works as expected but only one heater used, useful for diamond extruders like hotends.
Anyway don't expect answers from makerbase. They just copied the smoothieware board and uses the firmware as it is.
Be kind to community and ask gently. Sometimes, someone, will answer your questions.
Cheers

rogerwilco3, 2017-08-31:
Hello make sure the M code is configured properly  the code supplyed are 889 in a std setting  change this to 104 and 109.in the std file the link.to.tool is absent i only changed the right m-code in the following line 

#temperature_control.hotend2.thermistor_pin    0.25             # Pin for the thermistor to read
#temperature_control.hotend2.heater_pin        2.6             # Pin that controls the heater
#temperature_control.hotend2.thermistor        RRRF100K        # see http://smoothieware.org/temperaturecontrol#toc5
##temperature_control.hotend2.beta             4066             # or set the beta value
**#temperature_control.hotend2.set_m_code        884              # change this to 104**
**#temperature_control.hotend2.set_and_wait_m_code 889            # change this to 109**
#temperature_control.hotend2.designator        T1               #
you will find now that the led at d12 will light up when Extruder 2 is heating up

rogerwilco3, 2017-08-31:
just as addon to this

M104 Snn set temperature in degrees Celsius
M109 Wait until heated bed is up to temp. 

greetings Wilco

closed 2020-06-06 by makerbase-mks