makerbase-mks/MKS-SBASE #28: Set temperature changes when printing starts

Press1now, 2017-06-04:
When I set the hot end temperature to 220 (C) and the hot bed to 100 (C) my display and program will show 220/220 and 100/100 respectively. But as soon as i start printing the set temperature from the MKS Sbase board changes it to  220/205 and 100/065 causing it to trigger a HALT message. What is causing this ?

MD3DDelta, 2017-06-07:
You have to set the pid tuning , and check very carefull the connextion from the thermosthat ,( what gives the temp) , mine was just broken ,when got warm the error came like you ,
So i solderd all again and pidtunend , and the problem was gone .
pidtune look to smoothie page .
cheers mike

Press1now, 2017-06-20:
Issue resolved, i was using Cura 2, and it was slicing the temperature of 100 and 65 respectively. Uninstalled it and went with another program.

closed 2017-06-20 by Press1now