makerbase-mks/MKS-SBASE #29: no xy homming

nelujones, 2017-06-05:
the z homing is ok but its imposible to homing the x and y i havr try to change from NO to NC change the miscroswitc every thing( i belive ) any help will be apriciate.

pixel3design-hub, 2017-06-06:
post your config.txt

nelujones, 2017-06-06:
Thanks for your rapid response. Was my error with the end stops. X end stop inverted with the Y and when the printer want to reach the x stops wasn't able beacouse was the Y stop so this it's ok for now the printer works perfectly. But right now I have another problem. Today I was getting at last my 3,2 tft mad trying to conect it to the mks sbase it's imposible. The baut rate same with the board, selected smoothie in the LCD firmware, las firmware uploaded to the LCD, cable is good tested with an tester what could be the problem ????

pixel3design-hub, 2017-06-06:
Si preferieres escribir en castellano seria mejor. No he entendido tu problema.
El TFT necesita ser configurado al igual que la placa madre

nelujones, 2017-06-06:
Jejejeje pues castellano entonces. Aver se k hay k programar la tft tengo ya una 3,2 con Ramps y va perfecto es una cosa rara ya k con Ramps no se comunica muchas veces aún k sean los mismos bautios. Ósea la anterior me va perfecto con Ramps y como me gustó tanto digo ahora compro otra y con su placa mks sbase v1.3 y su tft 3.2 pues no hay manera de k se comuniquen. Mismos bautios en el firmware seleccionado placa smoothie opción 3 creo verificado 50 veces el cable pues no hay manera. Me tocara intercambiar las tft aver el resultado ya k no se el por k. Algún idea ?

pixel3design-hub, 2017-06-07:
Esperate... pero tiènes la versiòn smoothie o la versiòn con el Mega2560?
Ya que son bastante distintas y en la segunda no puedo ayudarte.
Si pero tiènes la smoothie (bueno, la còpia) entonces puede haber dos problemas. El primero es que no tiènes que habilitar ningùn panel en la configuraciòn de la MKS-SBASE, sòlo poner:
panel.enable								false
y luego, un vèz prendida, ir a > panèl => Set => Connect => 250000 (yo la tengo en 250000, pero tiènes que ponèr la velocidad al iguàl que la tiènes en el config.txt de la placa SBASE.

No es la placa que controla el TFT si no al revès. El TFT es una mini computadora que controla a travès de la puerta de sèrie la SBASE. Se conecta de forma distinta en la RAMPS, ya que necesita una fuente de alimentaciòn a parte. En la SBASE NO LA NECESITA y se debe cambiar el jumper en la TFT y conectarla en AUX1 de la SBASE misma...
Me he hecho un lìo... Lo siento. Espero que hebas entendido. Yo soy italiano aùn que he vivido casi sei años en Motrìl, asì que perdòna mi castellano...
Un saludo y si tienes otras preguntas, intentarè de ayudarte.

nelujones, 2017-06-07:
Muchas gracias lo de los puertos lo tenía claro y lo hice cambie por la vieja tft 3.2 de mi primera impresora y funciona perfectamente puse la nueva tft3.2 y no funciona hice un vídeo al vendedor de bangood y me envía una nueva por lo visto es un problema de hardware. Así k muchas gracias por tu tiempo. Una cosita tú utilizas la smoothie me podrías mandar o poner tu configuración aquí. Gracias amigo.

MD3DDelta, 2017-06-07:
right again , englisch please so every body can read what you do and so .

nelujones, 2017-06-07:
MD3DDelta sorry for the Spanish language. Was more easy for me explaining the problem to pixel3design. But thanks was my error for not could be able to do homing on my printer was inverted the end switch x with y so when trying to do a homing the z axis god and when look to encounter the x end stop was not able to do homing. For now the problem solved. Thanks for your replay.

closed 2020-06-06 by makerbase-mks