makerbase-mks/MKS-SBASE #32: Viki 2 LCD wiring

scspieker, 2017-07-08:
Has anyone successfully wired the viki 2 LCD panel to the Sbase board?  I have been following examples and the docs from panuchette and have had varying success with lights and power.  I have NOT burned out the LCD with miswiring as I have another test bed working to confirm the state of the panel.

Any assistance would be greatly appreciated!

pixel3design-hub, 2017-07-17:
Unfortunately the board have the 2 connectors physically turned by 180deg.

scspieker, 2017-07-17:
I am not sure what you mean here - the connectors that I have are a solid dupont type at the LCD side while the  MKS side is a series of single ended connections.  This means that the MKS side is configured to allow any combination of to any pins necessary.  I just need to know which pins from the Viki 2.0 go to which pins on the MKS board.  I have had the LCD light (with no text), the LED Ring go from Blue to Red after boot and the reset button perform its behavioral action.  When the circle button is pushed there is a slight buzz - so I am sure that I have those worked out.  I have the LCD brightness at 8 and contrast setting at 4 but no matter what setting I have those two at the LCD doesn't seem to change.

Looking at the smoothieware page gives Azteeg x5 settings and 4 and 5 stepper configs - neither of these configurations seem to make any change when wired as recommended.

I was hoping that someone may have actually had this working and could describe and/or show what those pinouts might be set.

pixel3design-hub, 2017-07-17:
[Wiring Diagram](https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/1043587235/original/Viki2_Wiring%20v1.2.pdf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJ2JSYZ7O3I4JO6DA%2F20170717%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170717T194402Z&X-Amz-Expires=300&X-Amz-Signature=bb4c2e6fa85e2e7ca001ae782964dcf13f51cb224a092268dd1d5318db3e1961&X-Amz-SignedHeaders=Host&response-content-type=application%2Fpdf)

At page 4 there is the Ramps 1.4 wiring.

[RAMPS 1.4](http://reprap.org/mediawiki/images/f/f6/RAMPS1.4schematic.png)

In the lower right corner, you can find the connector and the Pin name shown in the previous pdf.

[RAMPS Adapter](http://photobucket.com/gallery/user/janwillem_01/media/cGF0aDovZzNkIHBpbm91dC5qcGc=/?ref=)
[RAMPS Adapter schametic](http://reprap.org/mediawiki/images/b/b2/RAMPS_GADGETS3_SHIELD_SCHEMATIC.jpg)
And there we are... Now you have all the information.

Next time you need help, instead of answering like that, try to find your way through St. Google.

closed 2020-06-06 by makerbase-mks