makerbase-mks/MKS-SBASE #33: sbase 1.3 extra steppers

Tertius91820, 2017-07-12:
Hi Guys

I'm currently running a core XY with dual extrusion.
Is there a way to implement 4 extruders?

Maybe external stepper expander boards via free pins?

And if free pins, which will be usable for 2 external stepper boards?

Thank you

MD3DDelta, 2017-07-28:
Hello , i was looking for the same problem .
didn't find anything , when i find something i give you answer.
cheers mike

Tertius91820, 2017-07-28:
Awesome thanks!

Tertius91820, 2017-11-05:
I'm using a driver for the x and a driver for y. Then I'm using one for two steppers on z. I've configured two extruders on the two other drivers e0 and e1. Where would I pin up for two more extruders?. I don't need thermal readings on separate extruders as I'm using one hotend like the Prusa...

Tertius91820, 2017-11-05:
Thanks for the info the repository👍 

Also I'm using sbase 1.3 and a tft2.8.

ovisopa, 2019-02-11:
I'm also interested if I can have 3 more extruders, if I will also purchase MKSTFT will the LCD1 and LCD2 headers be free ? 

If there is anyone that already tried, or knows for sure, please let us know what are the free pins we can use to drive at least 3 extruders

[image]

closed 2020-06-06 by makerbase-mks