makerbase-mks/MKS-SBASE #34: Pause extruder or change filament problem

se3d, 2017-07-25:
Using a mks touch 3.2 display, the pause or change fillament when back (resume print),  the extruder motor freeze or goes to wrong direction.

salcofago, 2017-07-25:
i had same problem

pixel3design-hub, 2017-07-26:
to help you out, I need the TFT and SBASE configs.

se3d, 2017-08-21:
**_TFT:_**

#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:1

#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1

#language (Simplified Chinese:1; traditional Chinese:2; English:3)
>cfg_language_type:3

#whether set machine auto-off after print finishes(auto-off:1; NO:0)
>cfg_print_finish_close_Machine:0

#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10

#extruder number(one:1; dual:2)
>cfg_sprayer_counter:1

#whether has heated bed(YES:1; NO: 0)
>cfg_custom_bed_flag:1

#the max target Temp of sprayer and heated bed
>cfg_max_sprayer_temperature:270
>cfg_max_hotbed_temperature:150

#the level signal of outage detection module PB1(low level:0; high level:1)
>cfg_PB0_PB1_Level:1

#wifi mode(AP:1; STA:2)
>cfg_wifi_mode:1

#wifi name and password
>cfg_wifi_ap_name:MKS_wifi
>cfg_wifi_key_code:makerbase

#Dynamic access IP
>cfg_ip_dhcp_flag:1

#IP/mask/gateway
>cfg_ip_address:192.168.3.100
>cfg_ip_mask:255.255.255.0
>cfg_ip_gate:192.168.3.1

#the color of screen background
>cfg_background_color:0x000000

#the color of title text
>cfg_title_color:0xFFFFFF

#the color of temperature,fan,(etc) background
>cfg_state_background_color:0x000000

#the color of temprature,fan,(etc) text
>cfg_state_text_color:0xFFFFFF

#the color of file button
>cfg_filename_background_color:0x000000

#the color of file text
>cfg_filename_color:0xFFFFFF

#the color of background in printing 
>cfg_printingstat_word_background_color:0x000000

#the color of text in printing status
>cfg_printingstat_word_color:0xFFFFFF
[... 374 lines cut ...]
# Example of a custom menu entry, which will show up in the Custom entry.
# NOTE _ gets converted to space in the menu and commands, | is used to separate multiple commands
custom_menu.power_on.enable                false              #true
custom_menu.power_on.name                  Power_on          #
custom_menu.power_on.command               M80               #

custom_menu.power_off.enable               false              #true
custom_menu.power_off.name                 Power_off         #
custom_menu.power_off.command              M81               #

# Only needed on a smoothieboard
currentcontrol_module_enable                  true             #

return_error_on_unhandled_gcode              true            #false

# network settings
network.enable                               true             # enable the ethernet network services
network.webserver.enable                     true             # enable the webserver
network.telnet.enable                        true             # enable the telnet server
network.plan9.enable                         true             # enable the plan9 network filesystem
network.ip_address  auto                     # use dhcp to get IP address
network.ip_address                           192.168.1.254    # the IP address
network.ip_mask                              255.255.255.0    # the ip mask
network.ip_gateway                           192.168.1.1      # the gateway address
network.hostname   King_Core                 # optionally set this hostname for dhcp

se3d, 2017-08-21:
[config.txt](https://github.com/makerbase-mks/MKS-SBASE/files/1237563/config.txt)
[config.txt](https://github.com/makerbase-mks/MKS-SBASE/files/1237565/config.txt)

pixel3design-hub, 2017-08-21:
X and Y Pos could not be -1, is invalid...
I got mine with X = 0 and Y = 250 (this must be relative to your machine bed dimensions)
Config Firmware must be set to 3, you are using a Smoothie knockoff

se3d, 2017-08-28:
Ok, I will try and post results. Thanks

se3d, 2017-09-12:
Ok. I got my code with your recommendations, but the same problem . wrong direction of extrusion, Z without touch the nozzle and no filament extrusion.

michail58, 2018-12-16:
У МЕНЯ ТАЖЕ ПРОБЛЕМА

closed 2020-06-06 by makerbase-mks