makerbase-mks/MKS-SBASE #36: Board is neither updating firmware nor responding on USB

mtavenrath, 2017-09-07:
What's a good way to get diagnostics from the board?

When powering the board with a USB card two files, firmware.bin and config and powering the board on D7 turns on. After around a minute the other 4 LEDs start to glim. The board also doesn't show up as USB device on a PC and doesn't react to any commands from the MKS TFT.

Looking at smoothieware.org one might want to try to update the bootloader, but it's not clear how to enter the rescue mode since there's no information which pins would have the RS232 pins for the bootloader flash and where the Bootloader (ISP) pin would be.

makerbase-mks, 2017-09-14:
After the board update firmware ,name of firmware.bin will rename the FIRMWARE.CUR. 
If you want to update the bootloader, you can update it in the J2 interface.

mtavenrath, 2017-09-14:
@makerbase-mks The board update through the sd card does not work. D7 stays green and the file never gets renamed.

Where can I can find a manual on how to flash the board through the J2 interface?

MD3DDelta, 2017-10-05:
Hi , in my case the mks tft works perfect ,when the board has 12volt powered on .
when power is off it has the problem what you have . 
format the sd-card and start over again , sometimes there are trouble with some sd-cards ,
use an small one . the new cards are sometimes to fast, the board works then halfway .
i use 8GB 10mb /s, that works .
greetz mike

pixel3design-hub, 2017-10-05:
https://github.com/pixel3design-hub/MKS-SBASE-FULL-DOCUMENTED

I made a fork and reuploaded the schematics. I hope it could help.

Filcanna, 2018-11-26:
It would be very kind to have clear indication of the pins to which you attach the programmer ...

closed 2020-06-06 by makerbase-mks