makerbase-mks/MKS-SBASE #37: Is it possible to do bed leveling with nozzle touch

vido89, 2017-09-20:
Hi, I wont to buy 3KU delta printer which has bed levelling but it does it by nozzle touching the heated bed but I do not know can same thing be done with sbase 1.3 can I configure it to do same deb leveling ?

[GitHub](https://www.aliexpress.com/store/product/2015-3KU-Easy-Install-Metal-Structure-Delta-3D-Printer-with-One-Filament-and-LCD-for-free/1373025_32364813740.html)

pixel3design-hub, 2017-10-04:
Sure you can.

vido89, 2017-10-04:
Tnx @pixel3design-hub Do you know any documention on this topc ?

pixel3design-hub, 2017-10-05:
there are plenty of projects on how to do that.
The best I found is about a Piezoelectric sensor.
The drawback of nozzle touch sensor (electrically speaking) is that plastic can isolate the electric contact between nozzle and aluminium bed plate. If this occours the nozzle will crash hardly on the bed and may damage it.
Inductive sensors are better and more reliables and easy to install.
Piezo are good but suffers of lot of false reads...
It's your choice. Let me know.
Cheers.

closed 2020-06-06 by makerbase-mks