makerbase-mks/MKS-SBASE #40: Is it possible to update steps per mm on the LCD?

dojo2017, 2017-10-11:
I am looking into using an MKS SBASE V1.3 board with MKS 12864 TFT LCD touch screen for a 3D printer and I want to verify that I will be able to update the steps per mm settings directly on the LCD. Thanks!

JanN-H, 2017-11-07:
That doesn't work and wouldn't make sense either. Perhaps you want to change printing speed? That works with the 12864 lcd, that is not s touchscreen.

pixel3design-hub, 2017-11-07:
No much sense, but is indeed possible.
But you need to edit the config of the TFT and create a menu with some GCodes dat will modify the steps/mm in the EEPROM, followed by an M500 to save the config.
Remember that the CONFIG on the SBASE board IS NOT MODIFIED.

JanN-H, 2017-11-07:
Just found out that is even possible with the 12864 diplaycontroller to change steps per mm while it is running - stupid...

closed 2020-06-06 by makerbase-mks