makerbase-mks/MKS-SBASE #42: Network is not working

FlameFlameFlame, 2017-10-19:
Hello, I am using MKS-SBASE V1.3 with the lastest Smoothieware firmware. My network is not working. Here is my configuration file, I've enabled network there and set up a static IP. It hasn't been working when DHCP were enabled too.
[config.txt](https://github.com/makerbase-mks/MKS-SBASE/files/1398172/config.txt)

pixel3design-hub, 2017-10-19:
try to cleanup the file.
more than 132 characters per line can break the interpreter. Remove all the comments or at least split em multiline.
try to ping the mks ip.
look into the router page if a dhcp was spooled for mks mac address.

ClemensEy, 2017-11-22:
I have the same problem. Here is my serial output for this config setting. It seems to hang at "EMAC_INIT". Nothing works afterwards, no display, no sd-card mount. Without the network enabled these things work fine. 
                               
config.txt:
> network.enable                               true            # enable the ethernet network services

serial output:
> Smoothie Running @100MHz
> Build version: edge-ededcd5, Build date: Oct 13 2017 13:20:17, MCU: LPC1768, System Clock: 100MHz
> 5 axis
>       NOTE: No extruders enabled
>                                 EMAC_INIT

closed 2020-06-06 by makerbase-mks