makerbase-mks/MKS-SBASE #43: Smoothieware - upstream commit of fork

oddstr13, 2017-10-25:
What is the upsteam [Smoothieware](https://github.com/Smoothieware/Smoothieware) commit [this fork](https://github.com/makerbase-mks/MKS-SBASE/tree/master/English/Source) is based on?

pixel3design-hub, 2018-03-23:
that's not a fork
That's a stolen work.
MKS just take it without give any credit or support to Smoothieware project.

oddstr13, 2018-03-24:
@pixel3design-hub I am fully aware. I was however trying to be a little diplomatic.

It is however a fork, as it has a common code base and is now "maintained" as a separate project. The legalities & licensing does not interfere with this technical definition.

The reason I am asking, is that I would like to get a proper diff between the original Smoothieware project and this code.

pixel3design-hub, 2018-03-25:
Sure... But a "_fork_" is intended to be a "_copy_" to work on, to make it more functional or to be used as base for other projects.
Indeed, there's _no source code on MKS repo_. Just an old firmware.bin from the _edge branch_ and no commits whatsoever.
They've been even removed references from Smoothieware project site.
They have even copied a piece of the Smoothieware site.
**That spoken, I've even been too diplomatic.**
I've got no support from Smoothieware because I use an MKS board. Smoothieware is Opensourced, but open doens't mean "_it could be stolen without consequences_".
_MKS peoples gives no help_, just take a look at how many time I was the only one answering questions.
**And now threre's a bug on Smoothieware firmware affecting just the MKS board.**
Smoothieware guys wont's support or listen to me (and they've nothing to be ashamed). Period.
_I've found by myself the bug but I'll never do a commit on this branch or in the Smoothieware one_. An that's a fact. Nothing to do with SW guys, but I won't help MKS more than I've done 'till now. 
Shame on me.
Instead I'll help community on the HW side. But software? No way.

**_# No more gifts to MKS._**

Cheers dude. And if you wanna talk, I'm always there.

closed 2020-06-06 by makerbase-mks