makerbase-mks/MKS-SBASE #44: No power to heaters? bed or extruder

undertowx, 2017-10-30:
I am new to this but can't see why I am not getting power to the pins for the bed or extruder heaters?
this is my config:

## Temperature control configuration
# See http://smoothieware.org/temperaturecontrol

# First hotend configuration
temperature_control.hotend.enable            true             # Whether to activate this ( "hotend" ) module at all.
temperature_control.hotend.thermistor_pin    0.24             # TH2 Pin for the thermistor to read
temperature_control.hotend.heater_pin        2.7              # HEATER-1 Pin that controls the heater, set to nc if a readonly thermistor is being defined
temperature_control.hotend.thermistor        RRRF10K	        # See http://smoothieware.org/temperaturecontrol#toc5
temperature_control.hotend.beta             4066              # Or set the beta value
temperature_control.hotend.set_m_code        104              # M-code to set the temperature for this module
temperature_control.hotend.set_and_wait_m_code 109            # M-code to set-and-wait for this module
temperature_control.hotend.designator        T                # Designator letter for this module
temperature_control.hotend.max_temp         300              # Set maximum temperature - Will prevent heating above 300 by default
temperature_control.hotend.min_temp         20                # Set minimum temperature - Will prevent heating below if set

# Safety control is enabled by default and can be overidden here, the values show the defaults
# See http://smoothieware.org/temperaturecontrol#runaway
temperature_control.hotend.runaway_heating_timeout      900  # How long it can take to heat up, max is 2040 seconds.
temperature_control.hotend.runaway_cooling_timeout        0  # How long it can take to cool down if temp is set lower, max is 2040 seconds
temperature_control.hotend.runaway_range                20   # How far from the set temperature it can wander, max setting is 63Â°C

# PID configuration 
# See http://smoothieware.org/temperaturecontrol#pid
#temperature_control.hotend.p_factor         13.7             # P ( proportional ) factor
#temperature_control.hotend.i_factor         0.097            # I ( integral ) factor
#temperature_control.hotend.d_factor         24               # D ( derivative ) factor

temperature_control.hotend.max_pwm          255              # Max pwm, 64 is a good value if driving a 12v resistor with 24v.

# Second hotend configuration
#temperature_control.hotend2.enable            false           # Whether to activate this ( "hotend" ) module at all.
#temperature_control.hotend2.thermistor_pin    0.25           # Pin for the thermistor to read
#temperature_control.hotend2.heater_pin        1.23           # Pin that controls the heater
#temperature_control.hotend2.thermistor        EPCOS100K      # See http://smoothieware.org/temperaturecontrol#thermistor
##temperature_control.hotend2.beta             4066           # or set the beta value
#temperature_control.hotend2.set_m_code        104            # M-code to set the temperature for this module
#temperature_control.hotend2.set_and_wait_m_code 109          # M-code to set-and-wait for this module
#temperature_control.hotend2.designator        T1             # Designator letter for this module

#temperature_control.hotend2.p_factor          13.7           # P ( proportional ) factor
#temperature_control.hotend2.i_factor          0.097          # I ( integral ) factor
#temperature_control.hotend2.d_factor          24             # D ( derivative ) factor

#temperature_control.hotend2.max_pwm          64              # Max pwm, 64 is a good value if driving a 12v resistor with 24v.

##Heated bed temperature control configuration
temperature_control.bed.enable               TRUE             # Whether to activate this ( "hotend" ) module at all.
temperature_control.bed.thermistor_pin       0.23             # TH1 Pin for the thermistor to read
temperature_control.bed.heater_pin           2.5              # Pin that controls the heater
temperature_control.bed.thermistor           Honeywell100K    # See http://smoothieware.org/temperaturecontrol#thermistor
temperature_control.bed.beta                4066             # Or set the beta value
temperature_control.bed.set_m_code           140              # M-code to set the temperature for this module
temperature_control.bed.set_and_wait_m_code  190              # M-code to set-and-wait for this module
temperature_control.bed.designator           B                # Designator letter for this module

# Bang-bang ( simplified ) control
# See http://smoothieware.org/temperaturecontrol#bang-bang

temperature_control.bed.bang_bang           true           # Set to true to use bang bang control rather than PID
temperature_control.bed.hysteresis          2.0              # Set to the temperature in degrees C to use as hysteresis

## Switch modules
# See http://smoothieware.org/switch

# Switch module for Bed fan control
switch.fan.enable                            true             # Enable this module
switch.fan.input_on_command                  M106             # Command that will turn this switch on
switch.fan.input_off_command                 M107             # Command that will turn this switch off
switch.fan.output_pin                        2.4              # FAN PIN Pin this module controls
switch.fan.output_type                       pwm              # PWM output settable with S parameter in the input_on_comand
#switch.fan.max_pwm                          255              # Set max pwm for the pin default is 255

 switch.misc.enable                          true             # Enable this module
 switch.misc.input_on_command                M42              # Command that will turn this switch on
 switch.misc.input_off_command               M43              # Command that will turn this switch off
 switch.misc.output_pin                      .26              # Pin this module controls
 switch.misc.output_type                     digital          # Digital means this is just an on or off pin

## Temperatureswitch
# See http://smoothieware.org/temperatureswitch
# Automatically toggle a switch at a specified temperature. Different ones of these may be defined to monitor different temperatures and switch different swithxes
# Useful to turn on a fan or water pump to cool the hotend
temperatureswitch.hotend.enable              true            #
temperatureswitch.hotend.designator          TH3              # first character of the temperature control designator to use as the temperature sensor to monitor
 temperatureswitch.hotend.switch              misc            # select which switch to use, matches the name of the defined switch
 temperatureswitch.hotend.threshold_temp      50.0            # temperature to turn on (if rising) or off the switch
 temperatureswitch.hotend.heatup_poll         15              # poll heatup at 15 sec intervals
 temperatureswitch.hotend.cooldown_poll       60              # poll cooldown at 60 sec intervals

pixel3design-hub, 2017-10-31:
Take off all' comments. Over 136 chars per line is not allowed by the firmware.

pixel3design-hub, 2017-10-31:
TH3 is not a designator. T3 is, unless you have modified profiles in slicing software and changed the designator

undertowx, 2017-10-31:
No luck with the heatbed, got the hotend working.
this is my heatbed config? whats wrong?
##Heated bed temperature control configuration
temperature_control.bed.enable               TRUE             # Whether to activate this ( "hotend" ) module at all.
temperature_control.bed.thermistor_pin       0.23             # TH1 Pin for the thermistor to read
temperature_control.bed.heater_pin           2.5              # Pin that controls the heater
#temperature_control.bed.thermistor           Honeywell100K	  # See http://smoothieware.org/temperaturecontrol#thermistor
temperature_control.bed.beta                 4066             # Or set the beta value
temperature_control.bed.set_m_code           140              # M-code to set the temperature for this module
temperature_control.bed.set_and_wait_m_code  190              # M-code to set-and-wait for this module
temperature_control.bed.designator           B                # Designator letter for this module

# Bang-bang ( simplified ) control
# See http://smoothieware.org/temperaturecontrol#bang-bang

temperature_control.bed.bang_bang           true           # Set to true to use bang bang control rather than PID
temperature_control.bed.hysteresis          2.0              # Set to the temperature in degrees C to use as hysteresis

undertowx, 2017-10-31:
hot end working, but not the heatbed? here is the config.

##Heated bed temperature control configuration
temperature_control.bed.enable               TRUE             # Whether to activate this ( "hotend" ) module at all.
temperature_control.bed.thermistor_pin       0.23             # TH1 Pin for the thermistor to read
temperature_control.bed.heater_pin           2.5              # Pin that controls the heater
#temperature_control.bed.thermistor           Honeywell100K	  # See http://smoothieware.org/temperaturecontrol#thermistor
temperature_control.bed.beta                 4066             # Or set the beta value
temperature_control.bed.set_m_code           140              # M-code to set the temperature for this module
temperature_control.bed.set_and_wait_m_code  190              # M-code to set-and-wait for this module
temperature_control.bed.designator           B                # Designator letter for this module

# Bang-bang ( simplified ) control
# See http://smoothieware.org/temperaturecontrol#bang-bang

temperature_control.bed.bang_bang           true           # Set to true to use bang bang control rather than PID
temperature_control.bed.hysteresis          2.0              # Set to the temperature in degrees C to use as hysteresis

undertowx, 2017-10-31:
hot end working, but not the heatbed? here is the config.

##Heated bed temperature control configuration
temperature_control.bed.enable               TRUE             
temperature_control.bed.thermistor_pin       0.23             
temperature_control.bed.heater_pin           2.5              
#temperature_control.bed.thermistor           Honeywell100K	  
temperature_control.bed.beta                 4066             
temperature_control.bed.set_m_code           140             
temperature_control.bed.set_and_wait_m_code  190              
temperature_control.bed.designator           B                
# Bang-bang ( simplified ) control
# See http://smoothieware.org/temperaturecontrol#bang-bang

temperature_control.bed.bang_bang           true           
temperature_control.bed.hysteresis          2.0

pixel3design-hub, 2017-11-01:
#temperature_control.bed.thermistor Honeywell100K

Take off the #

undertowx, 2017-11-01:
still not sending any power to it!

closed 2017-11-03 by undertowx