makerbase-mks/MKS-SBASE #45: Sbase not recognized  anymore in windows 10

OneDayFlie, 2017-11-04:
Sbase not recognized  anymore in windows 10 

Was loading new configs and adding a tft32...
But my connection got lost...
Tried a new usb cable... Diffrent Microsd... Old config/firmware...
But nothing seems to help...

denisbrandao, 2018-03-02:
have you find any solution on this? I am with the same problem.

OneDayFlie, 2018-03-02:
I bought a second one...  And sended the faulty one back to china... Still waiting for a response 
Marlin 2.0 works on it... but  still no usb... 
I bought for my hevo a chinese Duet Wifi instead

denisbrandao, 2018-03-02:
I have an original Duet. it is amazing. And I managed to put the updated Soothie firmware on SBASE. worked for a bit and than got this.  cant even turn it on. :-/ Just the processor overheating

OneDayFlie, 2018-03-02:
They are realy sensitive to electro magnetic fields or close the webgui grrr...
 Not as secure as the Duet ...

OneDayFlie, 2018-09-23:
Second board has the same problem... 
MKS SBASE  is real junk like that...
Where is the response from them???

closed 2020-06-06 by makerbase-mks

OneDayFlie, 2020-06-06:
Never fixed the problem... so why close it????
That's not how it works guys.
Do properly job or buy my sbase board back