makerbase-mks/MKS-SBASE #47: Delta, have to re-home after every print?

undertowx, 2017-11-07:
Just got my delta up and running, I was wondering what I am missing. I have to re-home after every print or the nozzle runs into the bed? Even if I am just reprinting the same part over again.
Is it something missing in config or Cura settings?
Thanks

MD3DDelta, 2017-11-30:
hy , did you set the hight every time corrected ?
and then saved by M500, then it will remember it hight ,when not then you got what you have now?
greeting mike

undertowx, 2017-11-30:
MD3DELTA.
Yes everything is fine. it is just a setting I had incorrect. to turn of the power to the motors after print finnished, which losses home position. The Gamma value does not change though.

closed 2020-06-06 by makerbase-mks