makerbase-mks/MKS-SBASE #48: mks sbase hangs at "EMAC_INIT" with network enabled

ClemensEy, 2017-11-22:
Here is my serial output for this config setting. It seems to hang at "EMAC_INIT". Nothing works afterwards, no display, no sd-card mount. Without the network enabled everything works fine. (mac override also doesn't work, it gives me "Invalid MAC address: 45.E2.F9.4E.BD.49")

config.txt:

```
# network settings
network.enable                               true            # enable the ethernet network services
network.webserver.enable                     true             # enable the webserver
network.telnet.enable                        true             # enable the telnet server
network.plan9.enable                         true             # enable the plan9 network filesystem
network.ip_address                           auto             # use dhcp to get ip address
#network.mac_override		                     45.E2.F9.4E.BD.49   
```

serial output:

```
Smoothie Running @100mhz
Build version: edge-ededcd5, Build date: Oct 13 2017 13:20:17, MCU: LPC1768, System Clock: 100MHz
5 axis
NOTE: No extruders enabled
EMAC_INIT
```

Thor368, 2018-05-03:
I got two boards doing the very same. I traced out the layout and found that the 3V3 never reach the PHY. Instead it has ~1,6V at its VDD1A and VDD2A pins where it's supposed to be at least 3V. Bridging the 3V3 supply with a nearby cap which is connected to the PHY makes networking work like a charm. As the entire board is masked black and a four layer construction I can not defnitly say what went wrong at MKS but I guess they missed a trace.

adriancuzman, 2018-06-12:
@Thor368 can you point out where did you solder on the board? Please :)  Do you have a picture or a drawing on the schematic? I think I have the same issue... Not booting when ethernet is enabled and also the green LED is not blinking...

Thor368, 2018-06-12:
I added a bridge between the positive leads of the big cap on the left (C43) and the small one on the right (C46). That's it. For my boards at least.

[image]

adriancuzman, 2018-06-18:
Thank you very much! 👍

closed 2020-06-06 by makerbase-mks

Primer-Merc, 2020-11-15:
So makerbase closed the issue but didn't issue a statement if this issue was ever fixed?

eskeyaar, 2021-07-06:
Having a issue where, i get a Yellow steady LED , but when enabled for ethernet, it doesnt show green nor @net when entered in pronterface : No network found.
I tried to bridge those C46, and C43 with a lead, but all my LED lights getting off, so it means its sorting out board. 
Stuck at this problem. If anyone can help, that would be great!

Thanks