makerbase-mks/MKS-SBASE #50: MKS SBase V1.3; getting an RC servo to move?

HugoW1975, 2017-11-30:
Hi,

I have my system running, I can move all axis and home all axis, hotend works, all fine (everything via Pronterface, BTW, getting the TFT up is a side project...). But my Z-endstop is mounted on an RC servo and  needs to be extended prior to homing, and retracted afterwards. Those commands I give in the G-code. I added this to the config, between the lines anouncing the switch modules and the block for the PWM control of a fan:

"#Switch module for servo control
switch.servo.enable                            true             # Enable this module
switch.servo.input_on_command                  M280             # M280 S7.5 would be midway
switch.servo.input_off_command                 M281             # Same as M280 S0 0% duty cycle, effectively off
switch.servo.output_pin                        1.23             # Must be a PWM capable pin
switch.servo.output_type                       hwpwm            # H/W pwm output settable with S parameter in the input_on_comand
switch.servo.pwm_period_ms                    20               # set period to 20ms (50Hz) default is 50Hz"

Minus the "", of course.

I checked pin 1.23 is a PWM-able pin and it is not in use anywhere else in the config file. If I tell the set-up "M280 S5", it should turn fully one way. "M280 S10" fully the other way, and "M280 S7.5" would / should be dead in the middle. But the servo connected does not move. I checked it gets 5V power, I checked the servo works in an RC car and I switched between servos. But to no avail. Any idea what I do wrong? The only thing I can think of is that the PWM signal of the board is 3.3V, not 5V. But I feel a modern servo shouldn't really mind?

Thanks,

Hugo

MD3DDelta, 2017-11-30:
hy , maybe the Ground is not well? for check the kabels, put an LED on the connections, of the servo .
software looks good to me .
i use hall-o- endstops works best at my delta .
greetings mike

HugoW1975, 2017-12-01:
Thanks for your input, Mike, I've checked all wiring and I was trying to read PWM signal using an Adruino as a scope. But I gon no signal. But:
I have reloaded the firmware and now it works. Go figure... But I'm happy it does. At the moment my servo is retracted at S3.7 and extended at S10.1. Warning for users of digital servos (like me); if the signal to the servo is too low (in my case below 3.5) it starts turning uncontrolled. And thus winds up the switch wiring if you don't stop it in time. Be ware! This BTW is not a firmare issue but a digital servo issue. It could however be covered if the firmare would limit the Sxx command.
Cheers,
Hugo

MD3DDelta, 2017-12-01:
fine that it works , happy printing 

cheers mike

HugoW1975, 2017-12-01:
Thanks Mike. Well, almost printing. Can I post issues about the G32 command as well, here? On other forums I don't get help as I don't run a real smoothieboard. I'll post a new issue and hope for help...

Hugo

MD3DDelta, 2017-12-15:
Hugo can you close this ??. 
then it is easyer to find peopple who need help for me and others 
cheers mike

closed 2017-12-17 by HugoW1975