makerbase-mks/MKS-SBASE #51: G32 does not understand probe distance to nozzle

HugoW1975, 2017-12-01:
Hi,

I have a retractable sensor on my diy printer, which stick out 3.5mm beyond the nozzle when probing. So I wrote the Z-min endstop config:

"gamma_min_endstop                            1.28^            # Pin to read min endstop, add a ! to invert if endstop is NO connected to ground
#gamma_max_endstop                           1.29^            # Pin to read max endstop, uncomment this and comment the above if using max endstops
gamma_homing_direction                       home_to_min      # Or set to home_to_max and set alpha_max and uncomment the alpha_max_endstop
gamma_min                                    3.5              # This gets loaded as the current position after homing when home_to_min is set
gamma_max                                    200              # This gets loaded as the current position after homing when home_to_max is set"

And that works. When I homed Z with G28 Z0, it knows after homing it is 3.5 mm above the bed. So far so good.

Now I want to use G32 bed leveling, or however it is called. So I wrote:
"# Levelling strategy
# Example for 3-point levelling strategy, see wiki documentation for other strategies
leveling-strategy.three-point-leveling.enable         true        # a leveling strategy that probes three points to define a plane and keeps the Z parallel to that plane
leveling-strategy.three-point-leveling.point1         10.0,10.0   # the first probe point (x,y) optional may be defined with M557
leveling-strategy.three-point-leveling.point2         10.0,330.0  # the second probe point (x,y)
leveling-strategy.three-point-leveling.point3         265,170.0   # the third probe point (x,y)
leveling-strategy.three-point-leveling.home_first     false       # home the XY axis before probing
leveling-strategy.three-point-leveling.tolerance      0.03        # the probe tolerance in mm, anything less that this will be ignored, default is 0.03mm
leveling-strategy.three-point-leveling.probe_offsets  0,-40,-3.5  # the probe offsets from nozzle, must be x,y,z, default is no offset
leveling-strategy.three-point-leveling.save_plane     true        # set to true to allow the bed plane to be saved with M500 default is false"

But the 3.5mm distance doesn't seem to stick. I tried setting it to zero, no change. I tried setting it to 3.5 (without the minus), no change. I even get a response when leveling:
"WARNING: probe is not within tolerance: 3.519844"

How can I tell it, when G32-ing, that the probe is 3.5 mm away from the bed?

Cheers,

Hugo

MD3DDelta, 2017-12-01:
OH Yeah , that i had also as problem g32 does not allow the probe setting -z, like you do.
i had that isseu too , i do this ,: g32  z0,02, when finish m500.
then set Z to 0 . M306 Z0 , Save M500 . this is wath i do when i used the printer along time ago,about an week or so, depends what temps is printingsset is. 
So: G32 Z0,02 , M500 , M306 Z0, M500 done. you can print several times about 10times . G32 is more to adjust the towers to the bed system , this is usual done when the printer changed ,(bed,Hotend),or moved arround , best probing is when bed and hotend is on the right temps, when is hot metall reacts a bit.
Or you do the g32 like you done and start then G29 , and put the code from probe at it.and start printing. maybe that work.
i have done this all with pronterface , here you can config buttons, works faster.
cheers mike

HugoW1975, 2017-12-02:
Thanks. I considered the M306 way but hoped I messed up somewhere... I'll go that route, but I don't M500. I recalibrate at every print, it's quick and easy and I never forget as it is in the Cura machine settings.

Cheers,

Hugo

MD3DDelta, 2017-12-15:
Hugo can you close this ??. you found the solution?? , then help others here by tell your solutions .
cheers mike

HugoW1975, 2017-12-17:
The only solution I found so far is to just G92 it to the correct position after G32-ing. Crude but effective...

closed 2017-12-17 by HugoW1975