makerbase-mks/MKS-SBASE #52: having issues with auto bed levelling.

foster1111, 2017-12-14:
it will not allow me to use a grid set up and the 3 point doesn't work particularly well

MD3DDelta, 2017-12-15:
Hy ,can you give more info????? your config ect . what kind printer ? 
and did you looked in the closed issues before? 

Then i can maybe help you , 
regards mike 
**mks - self , does not much support at all we help each other , i help here because some people helped me. ##

foster1111, 2017-12-15:
Thanks. its a standard cartesian, and what do you mean by closed issues. do you mean making sure that the end-stops are closed. basically i want to have the printer to probe the bed in a grid formation but the only one that it recognises is the 3 prob point it basically says that no strategy path can be found when i have the grid set up in the configuration file. so i have ended up using the 3 point probe, it probes the first area fine then goes to probe the second and just glides over doesn't probe and goes to home and continually spins. it feels like i have tried everything i have made sure that it probes in a triangle formation but still no luck. as you say there is not much support so iam struggling to find out what the issue is. thanks for your help

MD3DDelta, 2017-12-15:
Hy , i mean  Closed issues from this github, if you tab/click on open issues(32), then you see closed(20).
those what are closed they are solved , well most off them .
Yeah , you are in the right way , he is probing like an delta stile printer .
1 or 2 settings are wrong .
Set cartesian stile is on ,most in beginning from configfile. delta had tobe off or it will be on
then look for probe stile in qaudrant or  cartesian , i have an delta so mine  cartesian knowing is not very big,
but i am building at the time an a cartesian .
Or post here your configtxt and i will look what i can do . 
smoothiware pages are also good for looking answers , you have only to wath out for pins alligment with mksbase. i think i readed there something about your problem . BUT do not tell that you have an mks board ,most wiil not help you then .
http://smoothieware.org/
cheers mike

foster1111, 2017-12-15:
here is my config

# Smoothieboard configuration file, see http://smoothieware.org/configuring-smoothie
# NOTE Lines must not exceed 132 characters, and '#' characters mean what follows is ignored
## Robot module configurations : general handling of movement G-codes and slicing into moves

# Basic motion configuration
default_feed_rate                            4000             # Default speed (mm/minute) for G1/G2/G3 moves
default_seek_rate                            4000             # Default speed (mm/minute) for G0 moves
mm_per_arc_segment                           0.0              # Fixed length for line segments that divide arcs, 0 to disable
#mm_per_line_segment                         5                # Cut lines into segments this size
mm_max_arc_error                             0.01             # The maximum error for line segments that divide arcs 0 to disable
                                                              # note it is invalid for both the above be 0
                                                              # if both are used, will use largest segment length based on radius

# Arm solution configuration : Cartesian robot. Translates mm positions into stepper positions
# See http://smoothieware.org/stepper-motors
alpha_steps_per_mm                           160               # Steps per mm for alpha ( X ) stepper
beta_steps_per_mm                            160               # Steps per mm for beta ( Y ) stepper
gamma_steps_per_mm                           8000             # Steps per mm for gamma ( Z ) stepper

# Planner module configuration : Look-ahead and acceleration configuration
# See http://smoothieware.org/motion-control
acceleration                                 3000             # Acceleration in mm/second/second.
#z_acceleration                              500              # Acceleration for Z only moves in mm/s^2, 0 uses acceleration which is the default. DO NOT SET ON A DELTA
junction_deviation                           0.05             # See http://smoothieware.org/motion-control#junction-deviation
#z_junction_deviation                        0.0              # For Z only moves, -1 uses junction_deviation, zero disables junction_deviation on z moves DO NOT SET ON A DELTA

# Cartesian axis speed limits
x_axis_max_speed                             30000            # Maximum speed in mm/min
y_axis_max_speed                             30000            # Maximum speed in mm/min
z_axis_max_speed                             300              # Maximum speed in mm/min

# Stepper module configuration 
# Pins are defined as  ports, and pin numbers, appending "!" to the number will invert a pin
# See http://smoothieware.org/pin-configuration and http://smoothieware.org/pinout
alpha_step_pin                               2.0              # Pin for alpha stepper step signal
alpha_dir_pin                                0.5!              # Pin for alpha stepper direction, add '!' to reverse direction
alpha_en_pin                                 0.4              # Pin for alpha enable pin
alpha_current                                1              # X stepper motor current
alpha_max_rate                               30000.0          # Maximum rate in mm/min

beta_step_pin                                2.1              # Pin for beta stepper step signal
beta_dir_pin                                 0.11             # Pin for beta stepper direction, add '!' to reverse direction
beta_en_pin                                  0.10             # Pin for beta enable
beta_current                                 1.5              # Y stepper motor current
beta_max_rate                                30000.0          # Maxmimum rate in mm/min

gamma_step_pin                               2.2              # Pin for gamma stepper step signal
gamma_dir_pin                                0.20!             # Pin for gamma stepper direction, add '!' to reverse direction
gamma_en_pin                                 0.19             # Pin for gamma enable
gamma_current                                1.5              # Z stepper motor current
gamma_max_rate                               300.0            # Maximum rate in mm/min

## Extruder module configuration
# See http://smoothieware.org/extruder
extruder.hotend.enable                          true          # Whether to activate the extruder module at all. All configuration is ignored if false
extruder.hotend.steps_per_mm                    400           # Steps per mm for extruder stepper
extruder.hotend.default_feed_rate               600           # Default rate ( mm/minute ) for moves where only the extruder moves
extruder.hotend.acceleration                    500           # Acceleration for the stepper motor mm/sec²
extruder.hotend.max_speed                       50            # Maximum speed in mm/s

extruder.hotend.step_pin                        2.3           # Pin for extruder step signal
extruder.hotend.dir_pin                         0.22!          # Pin for extruder dir signal ( add '!' to reverse direction )
extruder.hotend.en_pin                          0.21          # Pin for extruder enable signal

# Extruder offset
extruder.hotend.x_offset                        17            # X offset from origin in mm
extruder.hotend.y_offset                        30            # Y offset from origin in mm
#extruder.hotend.z_offset                        0            # Z offset from origin in mm
[... 274 lines cut ...]
[... cut ...]
#msd_disable                                 false            # Disable the MSD (USB SDCARD), see http://smoothieware.org/troubleshooting#disable-msd
#dfu_enable                                  false            # For linux developers, set to true to enable DFU

# Only needed on a smoothieboard
# See http://smoothieware.org/currentcontrol
currentcontrol_module_enable                 true             # Control stepper motor current via the configuration file

any help would be greatly appreciated. thanks

MD3DDelta, 2017-12-15:
Hy foster 
 in ## here you have to find the answer in mine smoothie is it a bit different ,could be older version and has no carrtisian info only delta .
_## **Levelling strategy
Example for 3-point levelling strategy, see wiki documentation for other strategies
__#leveling-strategy.three-point-leveling.enable true_ # a leveling strategy that probes three points to define a plane and keeps the Z parallel to that plane_

i will look for you where maybe info is , stay tuned 
mike

MD3DDelta, 2017-12-15:
http://smoothieware.org/zprobe

here you find all the stuff 
mike

MD3DDelta, 2017-12-15:
http://forums.reprap.org/read.php?418,674718

This is also an nice info , if you have later problemms again 

cheers mike 
don't vergot to close the issue when you find the answers .

foster1111, 2017-12-16:
Cheers mike your help is greatly appreciated. ill closed the case if i sort the problem

closed 2018-05-29 by makerbase-mks