makerbase-mks/MKS-SBASE #53: retraction not working correctly

killisch, 2017-12-15:
I've found no matter what slicer i am using retraction results in the extruder pushing extra filament out for the extraction moving and continuing to print. anyone else had an issue like this? i have reversed the extruder direction in the config file as it was extruding filament the wrong direction initially.

MD3DDelta, 2017-12-15:
Hy ,can you give more info????? and did you looked in the closed issues before? Then i can maybe help you , regards mike **mks - self , does not much support at all we help each other , i help here because some people helped me. ##
--

i can sett this in slic3r in advanged /expert mode .others i don't know
regards mike

MD3DDelta, 2017-12-15:
sorry for thr big fonts something got wrong with copy txt

killisch, 2017-12-15:
All sorted thankyou .. guess if you put the ! mark before the value it doesn't want to work properly. managed to solve it after a bit of digging.. 
thanks :)

closed 2017-12-15 by killisch