makerbase-mks/MKS-SBASE #56: How to print from micro SD card?

HugoW1975, 2017-12-25:
Hi,

I have the MKS SBase V1.3 and it's TFT32 screen working. The machine is putting out nice prints, but it is running on Pronterface on my PC via the USB cable. I would like the machine to be stand alone and move it to my workshop. I can copy the gcode file to the Micro SD card which is in the Sbase, but I cannot find it via the screen. Also I can copy it to a normal SD card in the TFT32, but also then I cannot find the file. For the USB stick connector in the TFT32 same effect.

What am I doing wrong? It might be down to using the wrong format /file extension so the screen cannot find it. I would really like to print from the micro SD card in the SBase, because I could then hook the SBase up to the LAN and convey the files from the PC to the printer over the network.

Please advise,
Cheers,

Hugo

HugoW1975, 2018-01-11:
Not sure why I closed this, I still want to print from the SD card in the SBase, not in the TFT slot, as that is probably an issue contributing to my problems.

Thanks for advice,

Hugo

BurtisaWord, 2018-04-12:
Command is:
M32 XYZ.gcode

http://smoothieware.org/supported-g-codes

closed 2020-06-06 by makerbase-mks