makerbase-mks/MKS-SBASE #59: is it possible to replace µC on sbase v1.3?

TurboMOD, 2018-01-02:
Hi,
i got an issue with analog inputs on my mks sbase. One of them got connected to 24V by a mistake. Now i think about replacing the µC. But iam not sure how to get it running after that. Can i just flash it with the firmware.bin from a sd-card or is there somethinge else i have to do?

pixel3design-hub, 2018-01-03:
Go to smoothieware.org.
Read the Manual.
Cheers

<https://mailtrack.io/> Sent with Mailtrack
<https://chrome.google.com/webstore/detail/mailtrack-for-gmail-inbox/ndnaehgpjlnokgebbaldlmgkapkpjkkb?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality>

TurboMOD, 2018-02-14:
Hi, 
thx for your reply :)
After replacing the µC i wanted to flash the bootloader. I wanted to follow this guide:
http://smoothieware.org/flashing-the-bootloader?s[]=bootloader
iam not sure about that "6-pin connector noted as “Serial”" since J2 pins are:

GND  TRST
GND  TCK
3V3    TMS

google says those pins are used for "jtag" so i gues iam wrong here. Any idea?

pixel3design-hub, 2018-02-14:
RX0 and TX0 are connected to AUX1 in MKS SBASE boards.
If you take a look at the schematics, P0.3 and P0.2 are RX0 and TX0 respectively.
The 6 pin header is for JTAG programming and debugging.
Once you got the right pins, it's easy peasy lemon squeezy...
Cheers

closed 2020-06-06 by makerbase-mks