makerbase-mks/MKS-SBASE #60: Printer stops during long prints.

HugoW1975, 2018-01-11:
Hi,

I run the MKS SBase v1.3 board in combination with the TFT32 screen, I print from the SD in the screen slot. The machine is a CoreXY printer, home design, I run 400 steps per mm on X and Y axles. I run smoothie firmware and config files and the prints come out nice. The only issue is the printer stops on occasion during long prints. I have had it stop 3 times now in the last few days, once after 3 hours, once after 8 hours (96% done…) and once after 5 hours. All on different print jobs. The one that stopped after 3 hours I restarted from start and finished without issue after 4,5 hours.

The system just cuts power to the motors, after it stopped moving I can move the motors by hand. The screen shows no error messages, it actually thinks it is still printing. I read about issues with stopping during prints before, but those were solved by running the smoothie firmware instead of the MKS firmware. I am already running the smoothie firmware.

The controller is in a forced-vented case (airflow over top and bottom) and all parts, like the motors, stay cool. I don't use a sensor to check for fillament running out.

Please advise.

HugoW1975, 2018-01-12:
It stopped again today, different print file, but I noticed the last two stops were both at 53% into the print. I remember one stop at 96% but I don't know the others. They are all longer prints, BTW, 4+ hours.

picatostes, 2018-07-28:
I have exactly the same problem. Did you get it fixed?

GutoFroes, 2018-08-30:
yesterday my printer stopped at 70%. other times it prints bigger parts without problems. there really is no apparent reason for the problem.

evandene, 2020-02-24:
Yep toss the screen and install the good old 12864 screen.
Over there they have no clue about customer service. They simple show disrespect for any customer.
For to make the 12864 to work on your MKS Sbase v1.3 just cut away a notch out the EXT1 male connector FROM THE 12864 screen in order to make the pin adjustments for the EXT1 cable.

3Dangel, 2020-02-25:
Uh....... I did not need to cut ANY thing from my Sbase v1.3 to work with
the Reprap Full Graphic Discounted LCD!
Did I miss anything!?

makerbase-mks, 2020-02-29:
@3Dangel 
MKS SBASE EXP1/EXP2 interface the bayonet position is opposite to that of the public version, so you need to cut off the protruding part of the connection cable and connect it to the motherboard.

closed 2020-06-06 by makerbase-mks