makerbase-mks/MKS-SBASE #64: MKS sBase 1.3 - TFT32 not powering up

dareksilver, 2018-01-24:
Recently purchased a sbase 1.3 to replace my Base 1.5 (wanted to try a low cost 32-bit board and this seemed like a good one).  I was using a TFT32 on my Base1.5, but when I try to connct the TFT32 to the sBase - it does not power up or anything.  Plugging the TFT32 into the Base 1.5 has i still operating as expected.

Is there a setting I'm missing in the smoothieware config - or could this be a bad AUX1 port on the board (and thus need a replacement board)?

medicshort87, 2018-01-24:
Make sure you enable second uart in config for sbase

dareksilver, 2018-01-24:
medicshort87 - I have checked the config file, and second_usb_serial_enable has been set to true.

Sadly, it does not seem to be working.  

I scroll down and notice that in the # Panel section that the panel.lcd is set to smoothiepanel - should this setting be changed?  (I can't find it in the documentation sadly)

medicshort87, 2018-01-24:
I don't think I done anything with the actual lcd panel section since it isn't hooked to exp1 or 2. Baud rates could be off on panel or board config

dareksilver, 2018-01-24:
Yeah, I pre-configured the TFT32 to make sure it's using 115200 for the baud rate - though the display is working on a MKS Base 1.5 I'm running, so I'm pretty sure I've narrows it to hardware issues with the AUX port.

I'll contact the seller for a replacement.

medicshort87, 2018-01-24:
I have heard of some of the sbase having the tx rx  pins reversed.

dareksilver, 2018-01-24:
Yeah, that sounds like a manufacturer issue - and I don't have the tools/parts necessary to make a adaptor to test that.

Thanks very much for your help medicshort87

medicshort87, 2018-01-24:
No problem! You could test it with a breadboard.

rmhorwitz, 2018-02-03:
I have this setup.  The MKS TFT32 uses the 5V provided by your Smoothieboard.  UART has nothing to do about powering this device, NOTHING!!!!!!!!!  The MKS SBASE V1.3 uses a Schotky diode to provide isolation between USB VBUS and 5VDC.  This creates voltage drop so if you verify the the voltage between pin 1 and 3 and alternately 2 and 4 you should see 4.7VDC.  YOU WILL NOT SEE THIS VOLTAGE UNLESS YOU HAVE POWER APPLIED TO THE MAIN BOARD.  USB VBUS POWER DOES NOT COUNT!!!!

As for communications,  these devices rely on a NULL MODEM connection.  Which means you must connection the TFT32 Rx to the Tx on the Main board and the Rx to the TX.  If this is not clear to you, let me know, I will help you.

Rick

dareksilver, 2018-02-03:
!! I was starting to consider that also, until I noticed the swapped RX and TX pins.

So I have to connect the board to mains voltage in order to get power to the TFT32.   I thought it weird it powered the Base 1.5 & TFT32 through the USB, but I would get nothing on the sBase.

I will give it a go, thanks rmhorwitz!

PriyankaSShinde, 2018-03-26:
Hello darksilver, I have the same issue - TFT is not power up to mks Sbased v1.3 board.Can u please tell me, how you solved this problem?

Nawakland, 2019-10-19:
Hi there !

The USB 5v power supply is not enough.
You need an external power supply 12-24v for your MKS Sbase board to be able to power up the TFT28-32 screen. Without it there is no 5v coming from the AUX-1 connector pin. In other hand make sure that on your TFT8-32 screen the jumper cap is set for 5v use.

Hope this helps.

closed 2020-06-06 by makerbase-mks