makerbase-mks/MKS-SBASE #67: Dual Y-Axis with external driver LV8727

bertrandletot, 2018-02-20:
Hi, 

On a cnc machine used with the sbase v1.3 i need 2 y axis.
Stepper motor Nema23 270 oz 3A  is too big for the internal driver, so i have 2 external driver LV8727 from mks.
I try to do that : http://smoothieware.org/cnc-mill-guide#doubling-stepper-motor-drivers
I'm using the last firmware [from smoothieware](https://github.com/Smoothieware/Smoothieware/tree/edge/FirmwareBin)
Can i just link 2 external driver LV8727 to the pinout jc_m3 ?

Thanks,

Bertrand

pixel3Design, 2018-02-21:
I think this is an hardware related question.
You should ask on the MKS project repository.

bertrandletot, 2018-02-22:
Hi pixel3Design, 
But I thought I was on the MKS project repository.

Bertrand

pixel3design-hub, 2018-02-22:
So sorry... My bad... Too many repos to look for...
And yes you can indeed.
Connect GND, EN, STEP, DIR pins respectively and should work

closed 2020-06-06 by makerbase-mks