makerbase-mks/MKS-SBASE #68: Problem with latest Smoothieware firmware.

pixel3design-hub, 2018-03-16:
I found that latest version of precompiled smoothie firmware.bin in master branch is buggy on MKS SBASE boards.
Printing results in vertically skewed objects.

I've teared down all the machine, rebuild it from ground up, changed motors, belts, bushings, rods, ball bearings and plastic parts.

Same result. Swapped 3 boards. No luck.
Then I roll back to an earlier fw version and BANG. It works.
So then I swapped the MKS board with a Smoothieboard and it worked. So I get back to the new firmware release on the Smoothieware board and it worked. Back again on the MKS with new firmware and it failed. Back on the old firmare and it worked.
No changes made on the configuration file whatsoever.

Seems that the BETA is losing steps or ALPHA does not have enough holding retention power.
Maybe a bug in the current control drivers code?

My machine is a CoreXY mechanism.

pixel3design-hub, 2018-03-25:
I found the bug. Resolved.
Thanks for NOTHING MKS

closed 2018-03-25 by pixel3design-hub

emavap, 2018-03-25:
Can you please let me know how did you resolve the bug?

phylyprobert, 2018-08-28:
Hey @pixel3design-hub this is phylyp from Brazil. Im build a 1x1x2m giant printer with tb6600 drivers and nem 23 steppers... and I got your same issue... can you email me at phylyp.robert@gmail.com with some solutions please? Thank you for now.

arnljot, 2018-10-31:
@pixel3design-hub, can you also email me? arnljot at gmail?

aceranic, 2018-11-16:
@phylyprobert , can you post a image of failed print