makerbase-mks/MKS-SBASE #70: Printer shuts down hot-end temp does not stabilize

rmhorwitz, 2018-03-20:
I am having grief with my new delta printer where I warm up the bed, than dial the hot end to 200 deg C.  Once it hits 200, the machine homes and proceeds to print.   While printing I see the temperature rise to 211-212 when the machine suddenly shuts down.  Any ideas?

Thanks,

Rick

pixel3design-hub, 2018-03-20:
Please post your config.

rmhorwitz, 2018-03-20:
Pixel- thank you in advance!!!
temperature_control.hotend.enable            true             # 
temperature_control.hotend.thermistor_pin    0.24             # 
temperature_control.hotend.heater_pin        2.7              # 
temperature_control.hotend.thermistor        RRRF100K         # 
temperature_control.hotend.set_m_code        104              # This allows us to set temperature using G-code m104
temperature_control.hotend.set_and_wait_m_code 109            # This allows us to wait until set temperature is met prior to continuing (G-code m104)
temperature_control.hotend.designator        T                # Gives the hotend the designation of 'T'
temperature_control.hotend.readings_per_second		20	# not sure if this will work
temperature_control.hotend.runaway_heating_timeout	120   # I give this 120 seconds to reach set temp
temperature_control.hotend.runaway_range		10    # If set to non-zero, and the target temperature is reached, and temperature diverges from the target temperature by more than this, the system will be forced into a HALT state.
temperatureswitch.hotend.enable		     true
temperatureswitch.hotend.switch              fan        #
temperatureswitch.hotend.threshold_temp      50               #  Turns ON Fan above this temperature, and OFF below this temperature.

#temperature_control.hotend.p_factor         13.7             # permanently set the PID values after an auto pid
#temperature_control.hotend.i_factor         0.097            #
#temperature_control.hotend.d_factor         24               #

# Z-probe uses the min stop for input and J8 port to output
zprobe.enable                                false           # set to true to enable a zprobe
zprobe.probe_pin                             1.28^          # pin probe is attached to if NC remove the !
zprobe.slow_feedrate                         5               # mm/sec probe feed rate
zprobe.fast_feedrate                         100            # move feedrate mm/sec
zprobe.probe_height                          5               # how much above bed to start probe
leveling-strategy.delta-calibration.enable   true            # basic delta calibration
leveling-strategy.delta-calibration.radius   100             # the probe radius

#---------------------------------------------------------------------
#?????????,????????

mm_per_arc_segment                           0.1              # Arcs are cut into segments ( lines ), this is the length for
                                                              # these segments.  Smaller values mean more resolution,
                                                              # higher values mean faster computation, was 0.5

delta_segments_per_second                    180              # for deltas only same as in Marlin/Delta, set to 0 to disable
                                                              # and use mm_per_line_segment

planner_queue_size                           32               # DO NOT CHANGE THIS UNLESS YOU KNOW EXACTLY WHAT YOU ARE DOING
microseconds_per_step_pulse                  1                # Duration of step pulses to stepper drivers, in microseconds
base_stepping_frequency                      100000           # Base frequency for stepping, higher gives smoother movement
acceleration_ticks_per_second                1000             # Number of times per second the speed is updated

uart0.baud_rate                              115200          # Baud rate for the default hardware serial port
second_usb_serial_enable                     true            # This enables a second usb serial port (to have both pronterface
                                                              # and a terminal connected)
pause_button_enable                          true             # Pause button enable

#????
temperature_control.bed.enable               true            #
temperature_control.bed.thermistor_pin       0.23             #
temperature_control.bed.heater_pin           2.5              #
temperature_control.bed.thermistor           RRRF100K    # see http://smoothieware.org/temperaturecontrol#toc5
temperature_control.bed.set_m_code           140              #
temperature_control.bed.set_and_wait_m_code  190              #
temperature_control.bed.designator           B                #
temperature_control.bed.runaway_heating_timeout	120   # I give this 120 seconds to reach set temp
emperature_control.bed.runaway_range	     10	              #If temperature deviates more than this value, kill machine 	
#temperature_control.bed.bang_bang            false           # set to true to use bang bang control rather than PID
#temperature_control.bed.hysteresis           2.0             # set to the temperature in degrees C to use as hysteresis
                                                              # when using bang bang

# Fan Control
switch.fan.enable                            true             #
switch.fan.input_on_command                  M106             #
switch.fan.input_off_command                 M107             #
switch.fan.output_pin                        2.4              #
switch.fan.output_type                       pwm              # pwm output settable with S parameter in the input_on_comand input_on_comand 

# Modeling Fan Control
[... 11 lines cut ...]
alpha_homing_direction                       home_to_max      # Home up
alpha_max                                    0                #
[... cut ...]
alpha_trim                                   0                 # software trim for alpha stepper endstop (in mm)
beta_trim                                    0                 # software trim for beta stepper endstop (in mm)
gamma_trim                                   0                 # software trim for gamma stepper endstop (in mm)

#endstop_debounce_count                       100              # uncomment if you get noise on your endstops
pause_button_enable                          true             #

rmhorwitz, 2018-03-21:
I just noticed that I am ignoring bang bang control.  I thought this was default. If it is, does it also have a default hysteresis?

pixel3design-hub, 2018-03-21:
emperature_control.hotend.runaway_range	10

This is the maximum drift from the imposed temperature. You set 200 and got 211/212. It's a corret behavior to shut down the hotend, because the temp "runs away" from the target temp.

Did you set the correct P.I.D. values?
Have you performed an autotune PID with M303 E0 S200 and then M500 to save the new PID?

rmhorwitz, 2018-03-21:
"Did you set the correct P.I.D. values?
Have you performed an autotune PID with M303 E0 S200 and then M500 to save the new PID?"

No I didn't.   I am totally ignorant of these parameters.  
Thanks,

Rick

rmhorwitz, 2018-03-21:
Why can't we use Bang Bang?

pixel3design-hub, 2018-03-21:
P.I.D. stands for Proportional Integral Derivative and is a control loop feedback system widely used in engineering. Short: it makes your temperature stable, with no or very little drift from your chosen temp value.

BangBang mode is just heating your hotend with little control and a lot of drifting from target temperature. Instead PID manage the temperature to make it less fluctuating.

Try those two commands from the (e.g. Repetier Host) console:
M303 E0 S200

It'll take sometime to complete, cause it turns on (M303) the HotEnd (E0) to 200° (S200) Celsius and do some loops to evaluate the three values for P, I and D.
Normally it does 3 cycles.
When it finished, run this command:
M500

It'll save your configuration in a file in the uSD card, the override config.

Now you are ready to print without printer hangs.

Cheers.

rmhorwitz, 2018-03-21:
Pixel- You are the best!!  Thank you

rmhorwitz, 2018-03-22:
Pixel- Works like a champ!  During a dry run I set my temp to 200 and watch it rise.  It goes to 210 momentarily, but falls down to 200 and sticks there, doesn't budge.  I tried other setting, same thing.

closed 2018-03-22 by rmhorwitz