makerbase-mks/MKS-SBASE #71: no power from E1 pinnout?

undertowx, 2018-03-24:
everything was fine, print lifted from bed and looks like a bunch of plastic blobbed around nozzle, now I have no power coming from the P2.7-E1 port on the board?
I guess I could move to E-2 P2,6 but would lose my extra fan as well?
Help?

pixel3design-hub, 2018-03-25:
Can you please paste your config?

undertowx, 2018-03-25:
# Smoothieboard configuration file, see http://smoothieware.org/configuring-smoothie
# NOTE Lines must not exceed 132 characters, and '#' characters mean what follows is ignored
## Robot module configurations : general handling of movement G-codes and slicing into moves

################ARH DELTA PRINTER##################################

# Basic motion configuration
default_feed_rate                            18000            
default_seek_rate                            18000            
mm_per_arc_segment                           0.0             
#mm_per_line_segment                         5                
mm_max_arc_error                             0.01             
                                                              
                                                              
delta_segments_per_second                    100              
                                                              

# Arm solution configuration : Cartesian robot. Translates mm positions into stepper positions
# See http://smoothieware.org/stepper-motors
alpha_steps_per_mm                           160              
beta_steps_per_mm                            160              
gamma_steps_per_mm                           160              

# Delta configuration
# See http://smoothieware.org/delta
arm_solution                                 linear_delta     
arm_length                                   405.156          
arm_radius                                   180.840            

# Planner module configuration : Look-ahead and acceleration configuration
# See http://smoothieware.org/motion-control
acceleration                                 3000             
#z_acceleration                              0                
junction_deviation                           0.05             
#z_junction_deviation                        0.0              

# Cartesian axis speed limits
x_axis_max_speed                             30000            
y_axis_max_speed                             30000            
z_axis_max_speed                             30000            

# Stepper module configuration 
# Pins are defined as  ports, and pin numbers, appending "!" to the number will invert a pin

alpha_step_pin                               2.0              
alpha_dir_pin                                0.5!              
alpha_en_pin                                 0.4              
alpha_current                                1.8
alpha_max_rate                               30000.0          

beta_step_pin                                2.1              
beta_dir_pin                                 0.11!             
beta_en_pin                                  0.10             
beta_current                                 1.8
beta_max_rate                                30000.0          

gamma_step_pin                               2.2              
gamma_dir_pin                                0.20!             
gamma_en_pin                                 0.19             
gamma_current                                1.8
gamma_max_rate                               30000.0          

## Extruder module configuration
# See http://smoothieware.org/extruder
extruder.hotend.enable                          true          
extruder.hotend.steps_per_mm                    190        
extruder.hotend.default_feed_rate               600           
extruder.hotend.acceleration                    500           
extruder.hotend.max_speed                       50            

[... 370 lines cut ...]
# For communication over the UART port, *not* the USB/Serial port
uart0.baud_rate                              115200           

second_usb_serial_enable                     false            
#leds_disable                                true             
#play_led_disable                            true             

# network settings
network.enable                               false          # enable the ethernet network services
network.webserver.enable                     true             # enable the webserver
network.telnet.enable                        true             # enable the telnet server
network.plan9.enable                         false             # enable the plan9 network filesystem
network.ip_address                           auto             # use dhcp to get ip address

# Kill button maybe assigned to a different pin, 
# See http://smoothieware.org/killbutton
kill_button_enable                           false            
kill_button_pin                              2.12             

#msd_disable                                 false            
#dfu_enable                                  false            

# Only needed on a smoothieboard
# See http://smoothieware.org/currentcontrol
currentcontrol_module_enable                 true

undertowx, 2018-03-25:
I know there isnt a problem with the cartridge heater because I swapped out to the Fan wiring on the board and the heater warmed up when I enabled the fan pin.. I tested the P2.7 pin and no volts are going to it and led does not light?

undertowx, 2018-03-26:
was the thermistor.

closed 2018-03-26 by undertowx