makerbase-mks/MKS-SBASE #72: Problem with RepRap Discount smart lcd?

siblues, 2018-03-26:
I am using the 12864 smart lcd on a MKS v1.3.I am new to all of this and have never had to set up a printer controller.I am just trying to get the panel working before I start trying to set up the board for use.I am using a usb to power up the board and panel/lcd.I do not know if it needs to be powered by a psu in order for the panel/lcd to function properly.Here is my config setup so far I only changed the leveling and panel sections and added a custom menu.The  LCD powers on but does not display any text it is blank.The knob/encoder beeps when it is pressed bu the screen is blank.

My config

[config grid level and lcd.txt](https://github.com/makerbase-mks/MKS-SBASE/files/1846701/config.grid.level.and.lcd.txt)

pixel3design-hub, 2018-03-27:
don't know if it's your case, but AFAIK the two EXP1 & EXP2 have been placed turned 180°CW on MKS boards.
[EXP Connectors](https://drive.google.com/file/d/1LlnB5Pf5-WGBj6iMN_FiHVgEVf0d4sQG/view?usp=sharing
https://drive.google.com/file/d/1tbIESfJm76sxT8cShJpTWiNxdo9ATae4/view?usp=sharing)

As you can see the arrow pointing down on the left side, indicates the pin 1 of the connector.
In the other photo you see clearly the square pad that's indicating the pin 1.
The connector is turned by 180°. Your cables are marked on one side with a (normally) red line, which indicates the pin 1 of the ribbon cable.

The easiest approach for noobies is to grind the tab on the cable connectors to be able to put them in without recrimp everything.

Hope this would help.

Cheers

siblues, 2018-03-27:
``Thanks but mine are already flipped.I figured ou the problem anyway the contrast on the LCD was really bad and the pot wasn't enough to fix it.The pot on the bottom of the actual LCD that is sandwiched between the pbd and LCD needs to be adjusted but I can't get to it.

makerbase-mks, 2018-03-29:
@siblues  EXP1 should connect to EXP1,   EXP2 should be connect to EXP2.

makerbase-mks, 2018-03-29:
@siblues EXP1 should be connect to EXP1, EXP2 should be connect to EXP2.

waran14, 2018-05-27:
@siblues did you have to still flip the EXP 1, 2 connectors on the MKS SBASE board ? Also, did you make any other change to the config file ?

siblues, 2018-05-27:
I just flipped them on the back of the lcd and selected the glcd in the config. I gave up on that lcd and am currently using the 3.2 TFT

waran14, 2018-05-27:
alright. I tried swapping the cables and connector. The LCD is not even powering on. but the encoder wheel seems to do something. I am not sure at this point if this is related to pins on the LCD itself. But you mentioned something about the contrast not set correctly. how were you able to pry open the LCD without bending or breaking the pins soldered to the pcb ?

siblues, 2018-05-27:
I broke one lcd and decided to just go with the TFT.

waran14, 2018-05-27:
@pixel3design-hub do you have the Full Graphics (12864) or the standard graphics controller ? I am unable to find options to invert the encoder direction and the steps per rotation. 

Per rotation it is going 2 lines. Not sure what options to change. Here is my panel section

panel.enable                                 true             # Set to true to enable the panel code

# Example for reprap discount GLCD
# on glcd EXP1 is to left and EXP2 is to right, pin 1 is bottom left, pin 2 is top left etc.
# +5v is EXP1 pin 10, Gnd is EXP1 pin 9
panel.lcd                                   reprap_discount_glcd     #
panel.spi_channel                           0                 # SPI channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16              # SPI chip select     ; GLCD EXP1 Pin 4
panel.encoder_a_pin                         3.25!^            # Encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!^            # Encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^            # Click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31              # Pin for buzzer      ; GLCD EXP1 Pin 1
panel.back_button_pin                       2.11!^            # Back button         ; GLCD EXP2 Pin 8

panel.menu_offset                            0                 # Some panels will need 1 here

panel.alpha_jog_feedrate                     6000              # X jogging feedrate in mm/min
panel.beta_jog_feedrate                      6000              # Y jogging feedrate in mm/min
panel.gamma_jog_feedrate                     200               # Z jogging feedrate in mm/min

panel.hotend_temperature                     185               # Temp to set hotend when preheat is selected
panel.bed_temperature                        60                # Temp to set bed when preheat is selected

pixel3design-hub, 2018-05-28:
OMG... Not not not.
Is swapped pinwise, not connector wise... JC...

waran14, 2018-05-28:
@pixel3design-hub Is your SD card slot being read correctly ? I am having issues with my SD card being read. I tried 2 different cards. On the LCD menu, I have 2 options
1. EXT and 2. SD. So, am guessing firmware is doing okay

closed 2018-05-29 by makerbase-mks

Filcanna, 2019-01-18:
THIS WORKED FOR ME TO MAKE LCD RECOGNIZED BY THE BOARD
# Example for reprap discount GLCD
# on glcd EXP1 is to left and EXP2 is to right, pin 1 is bottom left, pin 2 is top left etc.
# +5v is EXP1 pin 10, Gnd is EXP1 pin 9
panel.spi_frequency        400000	
panel.lcd                                   reprap_discount_glcd  #st7565_glcd     #
panel.spi_channel                           0                 # SPI channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16              # SPI chip select     ; GLCD EXP1 Pin 4
panel.encoder_a_pin                         3.25!            # Encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!            # Encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^            # Click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31              # Pin for buzzer      ; GLCD EXP1 Pin 1
panel.back_button_pin                       2.11!^            # Back button         ; GLCD EXP2 Pin 8
panel.encoder_resolution                    4
panel.menu_offset                            0                 # Some panels will need 1 here

#panel.external_sd.spi_channel 		0 
#panel.external_sd.spi_cs_pin 		2.8
#panel.external_sd.sdcd_pin 		2.13!^