makerbase-mks/MKS-SBASE #76: e3d PT100 thermocouple and MKS SBASE v1.3

ulysse31, 2018-05-08:
Hi

Following your wiki, in order to link an e3d PT100 thermocouple to the Sbase, we need to connect the 5v pin of the thermocouple to the pin 2 of J2, which is suppose to be a 3.3v for SPI ...

BUT, where is that damn pin 2 of j2 ? (Sorry but I've been searching it everywhere ^^' )

Also, if it is only available on v1.2/1.1  (who knows?) I saw that their is a jumper set for powering sensors with 5v or 3.3v, which means 3 pins, one in 3.3v another in 5v, and the last to transfer that to the sensors. Since my sensors are in 5v, can I use the left 3.3v from that jumper to power my PT100 thermocouple ?

Thanks a lot,

Regards,

--
Ulysse31

ulysse31, 2018-05-09:
Answering my own issue : 

After receiving an answer from Makerbase directly, I received a PCB annotated image with the position of Pin 2 on J2. I also updated the wiki page on PT100, if someone could download the image, add it to the repo and update the link ... this upload won't last very long ...

My other question still pending : I saw that their is a jumper set for powering sensors with 5v or 3.3v, which means 3 pins, one in 3.3v another in 5v, and the last to transfer that to the sensors. Since my sensors are in 5v, can I use the left 3.3v from that jumper to power my PT100 thermocouple ?

anrual, 2018-05-22:
Hello 
Can you Copy de config file 

Thanks

evandene, 2019-01-15:
I would like to upgrade my MKS v1.3 Makerbase with two PT100 amplifiers for my dual extruder printer.
I understand now how to install one but can't get my head around how install two.
Thanks for helping me out some how.

pietrobaricco, 2019-02-08:
Hi! Yesterday I managed to have the pt100 working on the sbase, on Marlin but should be the same with smoothie: 
Connect the amplifier board(s) 5v to 3v3 and gnd and then use P0_02 and/or P0_03 as your analog pins. P1_30 and P1_31 are fine as well but you can't use them along with a discount lcd.
If powering the board with 3v3 you will have to tweak the thermistor table 20 scaling all temps to 66%.

devrdander, 2019-02-13:
> Hi! Yesterday I managed to have the pt100 working on the sbase, on Marlin but should be the same with smoothie:
> Connect the amplifier board(s) 5v to 3v3 and gnd and then use P0_02 and/or P0_03 as [...]

Pietro, what does your pins.h look like, does it accept p0_02 when configuring an analog input? What file/line did you edit to scale down all temps?

eulogix, 2019-02-13:
Currently running two trianglelab PT100s with two amplifiers, the relevant changes are

in pins_MKS_SBASE.h

    #define TEMP_0_PIN          6   // was 1  6 means P0_03 as ANALOG
    #define TEMP_1_PIN          7   // was 2, 7 means P0_02 as ANALOG

the caveat here is to refer to P0_02 as 7 (the index it has in the adc_pin_table array defined in the framework-arduino-lpc176x package in file pinmapping.h).
If you define TEMP_X_PIN as P0_02 directly, it doesn't work, I think because the pin gets configured as digital input or output).

About scaling the temps, I used a preprocessor macro in thermistor_20.h

    #define T20_ADC_SCALE(a) ((int)(0.86 * a))

     const short temptable_20[][2] PROGMEM = {
       { OV(  0),  T20_ADC_SCALE(0) },
       { OV(227),  T20_ADC_SCALE(1) },
       { OV(236),  T20_ADC_SCALE(10) }, 
       [...]

but the good news is that...it wasn't necessary! I suspected that the readings were off because I was comparing them to the values returned by a digital thermometer applied to the heatblock, which were significantly lower (33% or so), but then actual prints revealed that the readings were in fact correct, and the thermometer was reading low because of poor interface with the block. So to sum it up, leave thermistor tables alone and you'll be just fine.

pietrobaricco, 2019-02-13:
Currently running two trianglelab PT100s with two amplifiers, the relevant changes are

in pins_MKS_SBASE.h

    #define TEMP_0_PIN          6   // was 1  6 means P0_03 as ANALOG
    #define TEMP_1_PIN          7   // was 2, 7 means P0_02 as ANALOG

the caveat here is to refer to P0_02 as 7 (the index it has in the adc_pin_table array defined in the framework-arduino-lpc176x package in file pinmapping.h).
If you define TEMP_X_PIN as P0_02 directly, it doesn't work, I think because the pin gets configured as digital input or output).

About scaling the temps, I used a preprocessor macro in thermistor_20.h

    #define T20_ADC_SCALE(a) ((int)(0.86 * a))

     const short temptable_20[][2] PROGMEM = {
       { OV(  0),  T20_ADC_SCALE(0) },
       { OV(227),  T20_ADC_SCALE(1) },
       { OV(236),  T20_ADC_SCALE(10) }, 
       [...]

but the good news is that...it wasn't necessary! I suspected that the readings were off because I was comparing them to the values returned by a digital thermometer applied to the heatblock, which were significantly lower (33% or so), but then actual prints revealed that the readings were in fact correct, and the thermometer was reading low because of poor interface with the block. So to sum it up, leave thermistor tables alone and you'll be just fine.

petermccracken, 2019-11-27:
> Currently running two trianglelab PT100s with two amplifiers, the relevant changes are
>
> in pins_MKS_SBASE.h
>
> ```
> #define TEMP_0_PIN          6   // was 1  6 means P0_03 as ANALOG
> #define [...]

Hi Pietro, exactly what ground pin did you use on your SBASE ? Was it the GND on the thermocouple pins at top right of the board ? Thanks!

Br Peter.

pietrobaricco, 2019-11-28:
Hi, no, I've used the two gnd pins in AUX-1, but I think that any gnd pin will do.
I recently recompiled Marlin and noticed that my previous setup no longer worked, something must have changed in the LPC framework. I had to redefine the pins in pins_MKS_SBASE.h as follows

```C
[...]
#define TEMP_0_PIN          P0_03  // was 6
#define TEMP_1_PIN          P0_02  // was 7
[...]
```

now directly referencing the pins, and no longer their index in this array, which used to exist in the LPC framework:

```C
constexpr std::array<pin_t, 8> adc_pin_table {
  P0_23, P0_24, P0_25, P0_26, P1_30, P1_31, P0_03, P0_02
};
```

petermccracken, 2019-11-28:
> Hi, no, I've used the two gnd pins in AUX-1, but I think that any gnd pin will do.
> I recently recompiled Marlin and noticed that my previous setup no longer worked, something must have changed in [...]

Thanks!! Interestingly for the temp pins marlin now seems to use the format P0_03_A6, it compiles but have not tested it yet.

petermccracken, 2019-11-28:
I can confirm configuration of TEMP_0_PIN     P0_03_A6 was successful and the PT100 is working fine with my SBASE 1.2. Great little board.

closed 2020-06-06 by makerbase-mks