makerbase-mks/MKS-SBASE #80: Sbase1.3, bltouch, smoothieware. Oddly low current coming OUT of BLt

swatti87, 2018-05-29:
VERY odd issue. 

Connected BLt and tested it like in this video:
https://www.youtube.com/watch?v=zn_qLmcB_b0

Obvious difference being my Sbase and a 5v tiny bulb being used instead of a LED.

Using the M280 S5.5 command and pushing in the probe the bulb lights up VERY dim. 

M119 command WILL NOT detect the position of the probe-pin.
Using inverted command "!" the M119 says "probe:0" always.
Not using "!" the M119 command says "Probe:1" always.
Both regardless of where the probe-pin actually is.
Using "^" or a resistor seems to have no difference.

MY theory is the BLtouch is letting too little current/power thru and the board cant detect the amount its currently sending. Is there a pin that can detect smaller currents/power? 

Please help. I have no options left, i dont know what to try...

swatti87, 2018-05-29:
UPDATE: 

When triggering, the currency jumps, i counted as high as 3.4v BUT it then drops to 0.4... Most likely the board simply ignores a quick jump like that. It SHOULD leave it on, but why wont it?

swatti87, 2018-06-04:
"Partially" solved. Got rid of BLtouch. 

Last MKS i will ever buy.

kkszysiu, 2018-10-29:
Any solution here?

closed 2020-06-06 by makerbase-mks