makerbase-mks/MKS-SBASE #82: MKS Sbase 1.3 not connecting to Ethernet

mullenba, 2018-06-09:
I recently bought a 1.3 card and updated to the most recent stable firmware.  When I connect it to the network, the activity lights flash, but it doesn't appear on my DHCP clients list.  If it's not getting an IP I can't connect to it to use it.

Config is set properly ( as far as I know)
```
## Network settings.  See http://smoothieware.org/network
network.enable                               true            # Enable the ethernet network services
network.webserver.enable                     true             # Enable the webserver
network.telnet.enable                        true             # Enable the telnet server
network.ip_address                           auto             # Use dhcp to get ip address
```
What troubleshooting can I do, or things can I look for before just sending it back?

sdtag, 2018-06-10:
here are my network settings.

## Network settings
# See http://smoothieware.org/network
network.enable                               true            # Enable the ethernet network services
network.webserver.enable                     true             # Enable the webserver
network.telnet.enable                        false             # Enable the telnet server
# network.ip_address                           auto             # Use dhcp to get ip address
# Uncomment the 3 below to manually setup ip address
network.ip_address                           192.168.1.222   # The IP address
network.ip_mask                              255.255.255.0   # The ip mask
network.ip_gateway                           192.168.1.1     # The gateway address
#network.mac_override                         xx.xx.xx.xx.xx.xx  # Override the mac address, only do this if you have a conflict

I gave it a static ip of 192.168.1.222
I enter http://192.168.1.222 in my browser and there it is.
it works for me
disconnect from the network before printing - mine crashes

hermitcrabslab, 2018-07-20:
At least yours lights on activity leds, mine is total dead on LAN, was like that from day zero, looks like I got a not so quality tested board...

avion23, 2018-10-17:
I have the same problem. This is my config:

`
> network.enable                               true            # Enable the ethernet network services
> network.webserver.enable                     true             # Enable the webserver
> network.telnet.enable                        true             # Enable the telnet server
> network.ip_address                           192.168.2.2    # The IP address
> network.ip_mask                              255.255.255.0   # The ip mask
> network.ip_gateway                           192.168.2.1     # The gateway address
> network.mac_override                         D1.C8.A1.F1.AE.CB    # Override the mac address, only do this if you have a conflict`

It doesn't even boot without the .mac_override. The leds on all ethernet jacks stay silent. ethtool tells me the link is down. I've tried a raspberry pi and a router.

I've checked my board and I have the updated crystals, i.e. the flat SMD version.

avion23, 2018-10-19:
Someone seems to have it fixed in issue 48: https://github.com/makerbase-mks/MKS-SBASE/issues/48

The 3.3V supply was forgotten. Connect C43 and C46 for an easy fix

closed 2020-06-06 by makerbase-mks

otede, 2023-11-04:
I've got a LUME printer with MKS SBASE v1.3
Here's my default network config:
# network settings
network.enable                               true            # enable the ethernet network services
network.webserver.enable                     true             # enable the webserver
network.telnet.enable                        true             # enable the telnet server
#network.ip_address                           auto             # use dhcp to get ip address
# uncomment the 3 below to manually setup ip address
network.ip_address                           192.168.3.221    # the IP address
network.ip_mask                              255.255.255.0    # the ip mask
network.ip_gateway                           192.168.3.1      # the gateway address
#network.mac_override                         xx.xx.xx.xx.xx.xx  # override the mac address, only do this if you have a conflict

ipconfig after just connecting the ethernet cable to the computer:
   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Realtek Gaming 2.5GbE Family Controller #2
   Physical Address. . . . . . . . . : D8-...-A2
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::...17(Preferred)
   Autoconfiguration IPv4 Address. . : 169.254.249.150(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.0.0
   Default Gateway . . . . . . . . . :
   DHCPv6 IAID . . . . . . . . . . . : 316193729
   DHCPv6 Client DUID. . . . . . . . : 00-...-A2
   DNS Servers . . . . . . . . . . . : fec0:0:0:ffff::1%1
                                       fec0:0:0:ffff::2%1
                                       fec0:0:0:ffff::3%1
   NetBIOS over Tcpip. . . . . . . . : Enabled

ipconfig after manually setting 192.168.3.221 for IPv4:
   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Realtek Gaming 2.5GbE Family Controller #2
   Physical Address. . . . . . . . . : D8-...-A2
   DHCP Enabled. . . . . . . . . . . : No
   Autoconfiguration Enabled . . . . : Yes
   Autoconfiguration IPv4 Address. . : 169.254.249.150(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.0.0
   IPv4 Address. . . . . . . . . . . : 192.168.3.221(Duplicate)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Default Gateway . . . . . . . . . : 192.168.3.1
   NetBIOS over Tcpip. . . . . . . . : Enabled

ping 192.168.3.221 returns connection timed out (no response)

ping 169.254.249.150 returns 
Reply from 169.254.249.150: bytes=32 time<1ms TTL=128
Reply from 169.254.249.150: bytes=32 time<1ms TTL=128
Reply from 169.254.249.150: bytes=32 time<1ms TTL=128
Reply from 169.254.249.150: bytes=32 time<1ms TTL=128
but the LEDs on the MKS SBASE board ethernet socket do not blink.

I already bridged the two contacts on the PCB per https://github.com/makerbase-mks/MKS-SBASE/issues/48

I'm rather desperate to not have to run to the printer with an SD card tens of times per day :/

otede, 2023-11-04:
Do I post it as a new issue or is this one going to be re-opened?

3Dangel, 2023-11-05:
Try to join Makerbsse FB group

otede, 2023-11-08:
> Try to join Makerbsse FB group
> […](#)

My thoughts after checking out the FB groups: can a board be more dead? 😀

Quas7, 2023-11-08:
is your ethernet connection directly between MKS and PC or do you have a switch in between. Normally, you need a cross-over ethernet cable for 1-to-1 direct connection if you network card can not detect this.

[https://www.computercablestore.com/what-are-ethernet-crossover-cables#:~:text=An%20Ethernet%20crossover%20cable%20is,computers%2C%20routers%2C%20and%20networks.](https://www.computercablestore.com/what-are-ethernet-crossover-cables#:~:text=An%20Ethernet%20crossover%20cable%20is,computers%2C%20routers%2C%20and%20networks.)

otede, 2023-11-08:
> is your ethernet connection directly between MKS and PC or do you have a switch in between? Normally, you need a cross-over ethernet cable for 1-to-1 direct connection if you network card can not [...]

I have a modem router so I could do away without crossover. I monitored the network log after plugging in the cable. All quiet.

"Also, you are maybe running into a smoothieware (firmware) issue and not an MKS hardware problem."
I only found this https://forum.makerforums.info/t/network-problem/77058
and I give up at this point. Re-flashing/updating firmware is a no go for me. I had problems flashing even the manufacturer-supplied firmware update.

Thank you for your input, I really appreciate it @Quas7 Hit me if you have any further fool-proof ideas but firmware juggling is not for me. I already spent 3 times more time fixing the printer than actually printing.

otede, 2023-11-09:
Progress?
https://pastebin.com/JYuq0bcS

3Dangel, 2023-11-09:
I think guys need to know that this board has been discontinued.
All documentation should refer to the Smoothieboard.