makerbase-mks/MKS-SBASE #83: Unable to read external SD card

waran14, 2018-06-14:
Guys!

I am running smoothieware on my MKS SBASE v1.3. I ahve the RRD Full graphics Smart Controller hooked up to it. The LCD and click wheel are working fine. But the SD card is unable to detect/read anything.

Under "Play" in the main menu, I get 2 options - ext and sd. But when i open ext (or sd for that matter), the display does not read/list the files in the card.

Am I missing a setting here ?

Filcanna, 2019-02-04:
Also me... I tried with a different SPI channel and lowering SPI  frequency, but nothing displays, although a short lag indicates a try to read card directory

3Dangel, 2019-02-04:
Not sure if max. Spa city and FAT32 are to be considered.

Cheers

closed 2020-06-06 by makerbase-mks