makerbase-mks/MKS-SBASE #86: amplified e3d pt100 problems on mks sbase v 1.3

jernejski, 2018-09-08:
Hi
bare with me since I am a complete beginner on all of this stuff. I am trying to upgrade my 3D printer with MKS Sbase v 1.3 board, e3d v6 hot end and e3d pt100 with amplifier... 
I am finding very little almost no information how to go about hooking it up. 
Board works printer works with a stock e3d thermistor, but I simply can't find reliable information on how to connect it, and at this point, I am not sure if I am connecting it wrong or configuring it wrong or both. 
I have tried connecting 5v amplifier pin to pin 2 on J2, used GND pin on Thermocouple pinout just above pin 0.17 and signal pin to both 1.30 and 1.31. I tried some other combinations with the same result
The amplifier board appears to be working, at least I think so (based on measurement of signal wire with multimeter), but all I get is temperature 0. If my wiring is correct then I presume the issue is in my config. 
And I can't wrap my head around how to configure the thing. I found at least 10 different configurations and none of them worked for me. 
I went through smoothieware and a few other sites and I am probably missing something. It certainly isn't as straightforward as I thought it to be. 
As far as I know, I've tried all possible combinations in the configuration I could find, but apparently I don't know what I am doing... :)
Well, I am sure there is someone who managed to make it work... I hope and I ask that kind soul for advice.

Thanks for any help...

arnljot, 2018-11-19:
Using anything else than thermistors with smoothie-based boards (original smoothieboard included) is difficult.

There's a bug in the stable firmware that the ad-amplifier module and the pt100-amplifier module has that is only fixed in official edge firmware. So the first thing is to use the edge firmware from the smoothieware repo at github if you use anything else than thermistors.

Next is you need to set the 3v3 vs 5v header on the mks sbase motherboard. You'll find it below the endstops.

Running the pt100 board on 5v will according to smoothieware "kill the port" and it will need to be replaced. It also needs to be AVCC and AGND pins or 3v3. MKS hasn't documented which pins are AVCC and it's quite opaque which pins on the smoothieboard design are.

So, if your PT100 amplifier is alive you should be able to configure it like this:

> temperature_control.hotend.enable            true
> temperature_control.hotend.sensor            pt100_e3d
> temperature_control.hotend.e3d_amplifier_pin 1.30 # must be a free ADC pin, not a temperature input

Then looking at the amp orienting the PT100 connector to the left, the PT100 sensor should just plug in on that side, no polarity I think to think of. Then, the connector on the right hand of the amplifier board will have the lower pin to GND, middle pin to signal (for instance P1.23) and topmost pin to 3v3. Despite it being marked with 5v. But that's maybe a limitation by Smoothie and not the amp.

If I were you I'd pick up 3v3 from an unused endstop connector after setting the jumper to 3v3 instead of 5v.

Flash edge firmware. And then see what happens.

If you can read code, then the [PT100 module source code](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/PT100_E3D.cpp) is helpful to read. In the "UpdateConfig" method you see that the only configuration it concerns it self with is the pin to read. Then in "adc_value_to_temperature" you see how it figures out the temp value in celcius.

In the code for [TemperatureControl.cpp](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp) where the "PT100_E3D" instance is created in "load_config" you see what else of config it's interested in.

Comparing the edge branch to master you [spot the fix](https://github.com/Smoothieware/Smoothieware/commit/573ed7c0c719008075acb1d36850e718a2707372#diff-b16176e1a77bf2a9b130f465ebb37d33) that was put into edge in march this year.

I used the smoothie source code and other resources shared on reprap forum when I tried to get my MKS board to work with the AD8495 module for Thermocouple K type. But I couldn't get it stable. Edge made the sensor work better as it also pulled temp during IDLE (no heater activity), but still when PWM to heater was off the temp was "crashing" to 0 in software but the resistance from the Thermocouple was as expected and thus also the voltage from the AD8495 amp.

I suspect that smoothiware still has a lot of work to be done on the edge branch before PT100 and AD8495s are as mature and stable as thermistors are curently.

jernejski, 2018-11-24:
Alright... 
Thank you very much for your help. 
I think my first mistake was for not doing research on an MKS Sbase board I realized later that it is smoothie board knockoff. Well, that's what you get for not doing proper research. 
The board itself works OK with thermistor at least for now. I almost gave up on the darn thing and I was gonna use it for a bigger printer for just PLA and ABS, and upgrade to an original smoothie or duet maybe. But now I will definitely give it a go with PT100 on MKS and wary about an upgrade later, that is if I get some free time. 
Thanks for your time and effort, and I will post something if I am successful.

closed 2020-06-06 by makerbase-mks