makerbase-mks/MKS-SBASE #87: Endstop keeps triggered after homing

KuroiSabato, 2018-09-27:
Have a new issue that just can't fix.
my endstops (x,y,z) does not untrigger after homing the first time.

better explained: after turning the machine on; can check using M119:

KuroiSabato, 2018-10-01:
ok... I guess it's enough. Last MKS product ever.

Pixel3DesignITA, 2018-10-03:
are your endstops conneted to gnd?
are they simple pulse switches or the ones with 3 wires and monted on a pcb?

Try to remove the pullup ^ from the config of ONE switch and check again.
If the second option (switch over pcb) is true, then absolutely remove the pullup ^ frome config, because the red wire is connected to +5v (or 3.3v if selected on the board) and 2 pullups are not good anyway and can cause problems.
Switches MUST be connected:
ALPHA & BETA & GAMMA -> Normally Opened and pulled Up and connected between GND and SIG
alpha_min_endstop 1.24!^
beta_min_endstop 1.26!^
gamma_min_endstop 1.28!^

Check and write back.

Disclaimer:
I'm NOT an MKS people. I justa have various MKS boards for my printers. Just here to help.
Cheers!

Pixel3DesignITA, 2018-10-03:
Di nulla...

loefstrand, 2019-08-08:
Hi!
I found that R33, R34, R35, R39, R40 & R41 has a incorrect value (1K instead of 10K) on my MKS SBASE V1.3. This makes my optocoupler end stops fail. The optocouplers can't pull a correct 0V, instead I get ~1.1V due to the "strong" pullup resistors.

I have changed the resistors on my board with the correct value (10K) and now the end stops works :)

I saw that even early BOM:s from the source has the same incorrect value on the resistors. So please update your BOM with the correct values.

/// Marcus

closed 2020-06-06 by makerbase-mks