makerbase-mks/MKS-SBASE #90: motors make noise

diyquad, 2018-11-04:
Hi,
I recently move to mks sbase and smoothieware, and i have a problem configuring the motors.

when i try to move any axis, i got a high pitched sound. the sound continues even after the move of the motor end. I have to power off totally the printer to make it stop (i have waited several time few minutes, change when noise begin, never stop).

i follow several faq/forum post, but still not working no matter my changes.
Also note that motors are moving. just that the noise coming from them is impossible and strange that it continu even after the end of mouvement...

I tryed to change to following:

- alpha_current : from 0.5 to 1.6 doesn't change the problem - motors are rated 1.5A
(same for beta/gamma)
- microseconds_per_step_pulse  - value beetween 1 to 5, doens't change
- change a motor - no effect

I don't know if it's a problem with firmware or the board ? since i have tryed different config.txt changes, maybe the board? its brand new but never know...

My printer:
P3 steel + mks sbase 1.3 (last smoothieware firmware) + tft screen + e3d v6 clone

Thanks a lot for your help

Here is my config.txt important part

`default_feed_rate                            4000             # Default rate ( mm/minute ) for G1/G2/G3 moves
default_seek_rate                            4000             # Default rate ( mm/minute ) for G0 moves
mm_per_arc_segment                           0.5              # Arcs are cut into segments ( lines ), this is the length for
                                                              # these segments.  Smaller values mean more resolution,
                                                              # higher values mean faster computation
mm_per_line_segment                          5                # Lines can be cut into segments ( not usefull with cartesian
                                                              # coordinates robots ).

# Arm solution configuration : Cartesian robot. Translates mm positions into stepper positions
alpha_steps_per_mm                           160               # Steps per mm for alpha stepper
beta_steps_per_mm                            160              # Steps per mm for beta stepper
gamma_steps_per_mm                           5120             # Steps per mm for gamma stepper

# Planner module configuration : Look-ahead and acceleration configuration
planner_queue_size                           32               # DO NOT CHANGE THIS UNLESS YOU KNOW EXACTLY WHAT YOU ARE DOING
acceleration                                 600             # Acceleration in mm/second/second.
#z_acceleration                              60              # Acceleration for Z only moves in mm/s^2, 0 uses acceleration which is the default. DO NOT SET ON A DELTA
"acceleration_ticks_per_second                1000             # 1000 Number of times per second the speed is updated
junction_deviation                           0.02             # Similar to the old "max_jerk", in millimeters,
                                                              # see https://github.com/grbl/grbl/blob/master/planner.c
                                                              # and https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8
                                                              # Lower values mean being more careful, higher values means being
                                                              # faster and have more jerk
#z_junction_deviation                        0.0              # for Z only moves, -1 uses junction_deviation, zero disables junction_deviation on z moves DO NOT SET ON A DELTA
#minimum_planner_speed                       0.0              # sets the minimum planner speed in mm/sec

# Stepper module configuration
microseconds_per_step_pulse                  2                # Duration of step pulses to stepper drivers, in microseconds
base_stepping_frequency                      100000           # Base frequency for stepping, higher gives smoother movement

# Cartesian axis speed limits
x_axis_max_speed                             5000            # mm/min
y_axis_max_speed                             5000            # mm/min
z_axis_max_speed                             400              # mm/min

# Stepper module pins ( ports, and pin numbers, appending "!" to the number will invert a pin )
alpha_step_pin                               2.0              # Pin for alpha stepper step signal
alpha_dir_pin                                0.5!             # Pin for alpha stepper direction
alpha_en_pin                                 0.4              # Pin for alpha enable pin
alpha_current                                1.5             # X stepper motor current
alpha_max_rate                               5000          # mm/min

beta_step_pin                                2.1              # Pin for beta stepper step signal
beta_dir_pin                                 0.11!             # Pin for beta stepper direction
beta_en_pin                                  0.10             # Pin for beta enable
beta_current                                 1.5            # Y stepper motor current
beta_max_rate                                5000          # mm/min

gamma_step_pin                               2.2              # Pin for gamma stepper step signal
gamma_dir_pin                                0.20!             # Pin for gamma stepper direction
gamma_en_pin                                 0.19             # Pin for gamma enable
gamma_current                                1.5            # Z stepper motor current
gamma_max_rate                               300            # mm/min`

makerbase-mks, 2018-11-06:
Generally speaking, the problem that the motor sound is relatively large is generally several aspects.
1. The problem of the motor itself (if you change the motor of other parameters, the same is the case, then this possibility should be ruled out)
2. It is the problem of setting the parameter drive current setting. Generally, it should not exceed or be too close to the rated current. It should be no problem.
3. It is a motor drive problem, its subdivision is relatively low, the mute effect is not so high (you can connect an axis through an external drive, compare it to see if it is a motor drive problem, or other reasons) .

lezabour, 2018-11-07:
Hi 
Thanks for answer
1) i have try others motors, sound still here (on all motors)
2) i have change current. Normally, motors are rated 1.4V, but i try to give from 0.2 to 1.5... Only way to have "less" sound problem, is put 0.3-0.4v but motors are not moving properly...
3) will try that
thanks for your help

cunbal, 2018-11-12:
I have the same problem with my HS8401 motors. The only solution I found is to change step stick drivers or swap to another board.

closed 2020-06-06 by makerbase-mks