makerbase-mks/MKS-SBASE #91: Board freezes when extruder heat is turned on

HugoW1975, 2018-11-04:
Hi,

my board was functioning but now it has a problem. When the heat it turned on for the extruder, the board freezes. When I start the printer, I can move all axis from the TFT screen, perform home function, etc. As soon as I ask it to preheat, it stops working. The LEDs keep blinking away happily, but the motor-hum stops. The axis do not move anymore, the tempsensor reads zero degrees C on the screen, etc. When I restart the machine, same thing. Everything works until I heat the extruder. I can also run a full print sequence from the SD card as long as I take the heating lines out of the g-code file (and remove fillament because the extruder stays cold). My hotend is working OK, I measured it and tested it on direct power from the 24V power source, the sensor provides good readings.

So, something is amiss on the board. What could it be, where do I start looking?
Cheers,

Hugo

makerbase-mks, 2018-11-06:
This may be because the heating head failed to heat up, causing the motherboard to freeze. Is there any error return when the motherboard freezes? You can check if the thermal installation is secure and there is no loosening. Or set the heating time longer. The collective can refer to these settings, such as“Runaway detection”“Initial heat-up runaway detection”
http://smoothieware.org/temperaturecontrol?s[]=temperature&s[]=control&s[]=bed&s[]=set&s[]=code

HugoW1975, 2018-11-06:
The heater is in perfect working order. The system has worked for about a year (not counting the countless fails on longer prints). Today I hooked up another control system and the heater performed well. It is for me the final fail of this board, I'm tossing it out. Too many fails on longer prints (100% fails  on everything over 3 hours) and now this, I've had enough. I just want to print..

Hugo

HugoW1975, 2019-01-07:
It proved to really be a fault on the board. I readdressed the IO to a different port (routing E0 to where E1 should be, and physically connecting it there too) and now that works again. The Original E0 control is just dead on the board...

closed 2019-01-07 by HugoW1975

podarok, 2020-01-27:
I have MKS SBASE v1.3 and faced with same issue, but in my case changing from E0 to E1 didn't help.