makerbase-mks/MKS-SBASE #94: Marlin 2.0 MKS SBase 1.3; set driver current

HugoW1975, 2019-01-07:
Hi,

I would like to know how and where the current settings for the DRV8825 drivers are defined. I cannot find a default setting anywhere, I can find this on the configuration_adv.h. page but I cannot figure it out (I just don't understand it):

[IMG]http://www.modelbouwforum.nl/media/50095652_598514860601811_7200556317201661952_n.244408/full[/IMG] 

I changed line 581, the number of channels, to 5.
I changed line 584 to have 5 variables, and set the first two (my guess is X and Y motors) to 1.5A
Should I enable line 565?
Or should I enable line 580?
Or both?
I tried to include the library mentioned at line 580 using PlatformIO in Atom, I don't know if I succeeded.
Any help is welcome.

Cheers,

Hugo

zuki07, 2019-07-24:
This is how I have mine setup and seems to run good.  My current settings may be different than yours but this is what works good for me on my ender 3.  did you also uncomment #define DIGIPOT_I2C?  I'm also assuming that you added the drv8825 drivers in the configuration.h.

#define DIGIPOT_I2C
#if ENABLED(DIGIPOT_I2C) && !defined(DIGIPOT_I2C_ADDRESS_A)
  /**
   * Common slave addresses:
   *
   *                        A   (A shifted)   B   (B shifted)  IC
   * Smoothie              0x2C (0x58)       0x2D (0x5A)       MCP4451
   * AZTEEG_X3_PRO         0x2C (0x58)       0x2E (0x5C)       MCP4451
   * AZTEEG_X5_MINI        0x2C (0x58)       0x2E (0x5C)       MCP4451
   * AZTEEG_X5_MINI_WIFI         0x58              0x5C        MCP4451
   * MIGHTYBOARD_REVE      0x2F (0x5E)                         MCP4018
   */
  #define DIGIPOT_I2C_ADDRESS_A 0x2C  // unshifted slave address for first DIGIPOT
  #define DIGIPOT_I2C_ADDRESS_B 0x2D  // unshifted slave address for second DIGIPOT
#endif

//#define DIGIPOT_MCP4018          // Requires library from https://github.com/stawel/SlowSoftI2CMaster
#define DIGIPOT_I2C_NUM_CHANNELS 5 // 5DPRINT: 4     AZTEEG_X3_PRO: 8     MKS SBASE: 5
// Actual motor currents in Amps. The number of entries must match DIGIPOT_I2C_NUM_CHANNELS.
// These correspond to the physical drivers, so be mindful if the order is changed.
#define DIGIPOT_I2C_MOTOR_CURRENTS { .9, .9, .9, .9, .9, }  //  AZTEEG_X3_PRO

makerbase-mks, 2020-04-02:
@HugoW1975  You can refer to it

closed 2020-04-02 by makerbase-mks

evandene, 2020-12-05:
For me a slightly different settings in the configuration_adv.h worked out 100%
I have a delta printer D=300 H=450
MARLIN 2.0 bugfix from November 20, 2020
Board MKS Sbase v1.3
Home made ultralight duo floating extruder, 2 nozzles

//#define DIGIPOT_MCP4018             // Requires https://github.com/stawel/SlowSoftI2CMaster
#define DIGIPOT_MCP4451

#if EITHER(DIGIPOT_MCP4018, DIGIPOT_MCP4451)
  #define DIGIPOT_I2C_NUM_CHANNELS 5  // 5DPRINT:4   AZTEEG_X3_PRO:8   MKS_SBASE:5   MIGHTYBOARD_REVE:5

  // Actual motor currents in Amps. The number of entries must match DIGIPOT_I2C_NUM_CHANNELS.
  // These correspond to the physical drivers, so be mindful if the order is changed.
  #define DIGIPOT_I2C_MOTOR_CURRENTS { 1.000, 1.000, 1.000, 0.400, 0.400 } // MKS_SBASE

  //#define DIGIPOT_USE_RAW_VALUES    // Use DIGIPOT_MOTOR_CURRENT raw wiper values (instead of A4988 motor currents)

  /**
   * Common slave addresses:
   *
   *                        A   (A shifted)   B   (B shifted)  IC
   * Smoothie              0x2C (0x58)       0x2D (0x5A)       MCP4451
   * AZTEEG_X3_PRO         0x2C (0x58)       0x2E (0x5C)       MCP4451
   * AZTEEG_X5_MINI        0x2C (0x58)       0x2E (0x5C)       MCP4451
   * AZTEEG_X5_MINI_WIFI         0x58              0x5C        MCP4451
   * MIGHTYBOARD_REVE      0x2F (0x5E)                         MCP4018
   */
  #define DIGIPOT_I2C_ADDRESS_A 0x2C  // Unshifted slave address for first DIGIPOT
  #define DIGIPOT_I2C_ADDRESS_B 0x2D  // Unshifted slave address for second DIGIPOT
#endif