makerbase-mks/MKS-SBASE #95: reprap_discount_GLCD not recognized by MKS SBase

Filcanna, 2019-01-18:
THIS WORKED FOR ME TO MAKE LCD RECOGNIZED BY THE BOARD :D :D
Example for reprap discount GLCD
on glcd EXP1 is to left and EXP2 is to right, pin 1 is bottom left, pin 2 is top left etc.
+5v is EXP1 pin 10, Gnd is EXP1 pin 9

panel.spi_frequency 400000
panel.lcd reprap_discount_glcd #st7565_glcd #
panel.spi_channel 0 # SPI channel to use ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin 0.16 # SPI chip select ; GLCD EXP1 Pin 4
panel.encoder_a_pin 3.25! # Encoder pin ; GLCD EXP2 Pin 3
panel.encoder_b_pin 3.26! # Encoder pin ; GLCD EXP2 Pin 5
panel.click_button_pin 1.30!^ # Click button ; GLCD EXP1 Pin 2
panel.buzz_pin 1.31 # Pin for buzzer ; GLCD EXP1 Pin 1
panel.back_button_pin 2.11!^ # Back button ; GLCD EXP2 Pin 8
panel.encoder_resolution 4
panel.menu_offset 0 # Some panels will need 1 here

#panel.external_sd.spi_channel 0
#panel.external_sd.spi_cs_pin 2.8
#panel.external_sd.sdcd_pin 2.13!^

Note: spi clock frequency explicitly declared (on LPC1768 system clock is 96 MHz, in Smoothieboard LPC1769 is 120 MHz so I proportionally adapted SPI clock); ^ character removed from encoder pins setting (CW rotation moves menu down); resolution of encoder set to 4; my try to have an external SD card was aborted :(

makerbase-mks, 2019-03-21:
panel.enable                                 true             # set to true to enable the panel code
panel.lcd                                   reprap_discount_glcd     #
panel.spi_channel                           0                 # spi channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16              # spi chip select     ; GLCD EXP1 Pin 4
panel.encoder_a_pin                         3.25!^            # encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!^            # encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^            # click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31              # pin for buzzer      ; GLCD EXP1 Pin 1
panel.back_button_pin                       2.11!^            # back button         ; GLCD EXP2 Pin 8
panel.external_sd                     true              # set to true if there is an extrernal sdcard on the panel
panel.external_sd.spi_channel         1                 # set spi channel the sdcard is on
panel.external_sd.spi_cs_pin          0.28              # set spi chip select for the sdcard (or any spare pin)
panel.external_sd.sdcd_pin            0.27!^            # sd detect signal (set to nc if no sdcard detect) (or any spare pin)

panel.menu_offset                            1                 # some panels will need 1 here
panel.alpha_jog_feedrate                     6000              # x jogging feedrate in mm/min
panel.beta_jog_feedrate                      6000              # y jogging feedrate in mm/min
panel.gamma_jog_feedrate                     200               # z jogging feedrate in mm/min
panel.hotend_temperature                     185               # temp to set hotend when preheat is selected
panel.bed_temperature                        60                # temp to set bed when preheat is selected

This is my lcd configuration, it can be displayed normally.
Our configuration doesn't seem to be any different.
Can you check if there is something wrong?

DarkAlchy, 2019-08-04:
Did you get this to work?  From everything I am reading the issue is they have the SPI pins backwards and the only solution I have found is cutting wires and swapping them which is barbaric.  In Marlin we would just go to the pins.h file and all done.  The pins that need to be swapped are 0_18 with 1_18, 0_15 with 1_20 and 0_16 with 1_19.  Why can't we do this via a pins file?

3Dangel, 2019-08-05:
mine is tested and it is working.
did you download the recent FW?
no cutting wire nothing...

DarkAlchy, 2019-08-05:
I did as I had to since I only learned of this yesterday.  I get the classic blue lit screen and nothing on it and the only way to solve that is wire swapping since we don't have access to the pins file.

Filcanna, 2019-08-05:
You could recompile Smoothieware. If I’m not wrong there is something like a pins file. As an experiment I recompiled from source, on a Raspberry Pi 3B+ and it was fine and almost quick.

Inviato da me !

Filcanna, 2019-08-05:
I add that in my case I have had 3 Sbase, only one had absent text on display, whatever I did. Is dedicated to my delta, driven by RPi and a TFT 2.8”, so no need of a display 

Inviato da me !

DarkAlchy, 2019-08-05:
I was reading about that and they are such aholes and you must use this to compile etc... I just threw my hands up.  I am on Windows and what they demand is rough now Marlin's compilation technique makes sense.

As far as the pins I can only take one source for that info as no other site out there talks about it but the source that shows it working and the wiring hack job is https://www.thingiverse.com/thing:3608472

Filcanna, 2019-08-05:
The link you refer is showing an SKR board, not an SBase. Lately I've been able to have text displaying without mods, on Smoothie F/W :)

Filcanna, 2019-08-05:
And... I think that solving the problem of no functional display is simply a matter of asking Arthur Wolf, about how display pins are managed in his firmware

DarkAlchy, 2019-08-05:
I read that the wiring is the same on SKR and MKS but is different on the real Smoothie and that is why I posted that.  I do not have a MKS.

closed 2020-06-06 by makerbase-mks