makerbase-mks/MKS-SBASE #96: MKS-SBASE v1.3 not woking

MeikVr6, 2019-02-06:
Dear guys,
I have a following problem
My new board don't work. I have download and configuration the firmeware and config.txt from this site.
I have update the Firmeware over the a 2GB Micro SD-Card. In the SD-Card stand after the Update firmeware.CUR Data file.
What is my mistake?
[config.txt](https://github.com/makerbase-mks/MKS-SBASE/files/2835336/config.txt)
[firmware.zip](https://github.com/makerbase-mks/MKS-SBASE/files/2835342/firmware.zip)

Thanks for the Help

BorgMaster, 2019-02-06:
Same problem, nothing seems to move. Any ideas ?

MeikVr6, 2019-02-07:
Hello
I would like testing the MKS sbase with Marlin.

BR
Meik

makerbase-mks, 2019-03-21:
After updating the configuration file through the sd card, whether to ensure that the sd card is inserted on the motherboard every time the power is turned on. Because every time you boot the motherboard, you need to read the parameters in the configuration file from the sd card. When the motherboard is powered on, there is no SD card inserted or there is no configuration file in the card. The motherboard can't read the data, it can't be used.

closed 2020-06-06 by makerbase-mks